dusty-nv/jetson-inference #881: when resume pytorch-ssd, epoch update problem

ingbeeedd, 2021-01-04:
When I bring in a checkpoint model with a `--resume` option, it goes on epoch -1 just like that. When I look at the code or actually run it, it ignores the existing epoch and start from 0. So I added the next code, which causes initialization errors on the `lr_scheduler` side. Is there a problem?

`
timer.start("Load Model")
    if args.resume:
        logging.info(f"Resume from the model {args.resume}")
        last_epoch = int(args.resume.split('/')[2].split('-')[3])
        net.load(args.resume)
`

and I run the code 
`
python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=32 --epochs=60 --resume=models/fruit/mb1-ssd-Epoch-58-Loss-3.4249941666920978.pth
`

Thank you

dusty-nv, 2021-01-04:
The model epochs aren't saved inside the PyTorch checkpoint, so the code doesn't know which epoch it last ran.

If you are having problems, you can also try setting `--pretrained-ssd` argument to your model instead of the `--resume` argument.

ingbeeedd, 2021-01-08:
The question is, I loaded a model with a valuation epoch of 1.9, but the loss did not decrease from 1.9, but the loss rate was higher than before. Can you tell me why?

dusty-nv, 2021-01-08:
I'm not exactly sure - can you try the `--pretrained-ssd` way?

Bobby2090, 2021-01-11:
> 
> 
> The question is, I loaded a model with a valuation epoch of 1.9, but the loss did not decrease from 1.9, but the loss rate was higher than before. Can you tell me why?

try to use the --resume way, I had met the same question too.

ingbeeedd, 2021-01-11:
[image]

Could you look at the picture above? Is it loaded normally and the training is done properly?

dusty-nv, 2021-01-11:
It is hard to tell without seeing the validation loss at the end of the epoch, but it appears to have loaded your previous model OK and the losses are decreasing.

closed 2021-01-20 by ingbeeedd