dusty-nv/jetson-inference #885: Very weird problem with a sample code

neilyoung, 2021-01-05:
I was trying to use an RTSP capable camera in my network as input for the object detection. I came across the `video-viewer.py` sample and it took me just minutes to figure out, that I would have to help the `videoSource` to determine the right codec by providing that magic extra parameter `--input-codec`, which goes down the stack then:

`python3 video-viewer.py --input-codec=h264 rtsp://192.168.188.52:554/onvif1`

And tadaa --- video on the screen :)

Then it took me another two hours to figure out, why this RTSP video source does not work with the simplest object detection sample. Basically I merged the unchanged `my-detection.py` with the  unchanged `video-viewer.py` by applying these changes to `my-detection.py`:

```
import sys
camera = jetson.utils.videoSource("rtsp://192.168.188.52:554/onvif1", argv=sys.argv)      

```

No video view...

I first commented all definitions and references to `net`, but in the end I had to comment the entire 

`import jetson.inference
`

statement in order to make my RTSP video window appear. Bummer...

I see, that the GStreamer pipleline reaches "PLAYING" state, but the last trace is about an allocation of a RingBuffer consisting of 4 buffers, then nothing. No preview, no inference. But the captured frames drop in...

Strange enough the detection works with `/dev/video0` instead of the `rtsp` source, so it could be "something with the video".... But on the other hand: Not loading `jetson.inference` helps a lot...

Anybody having any idea, what's going wrong here?

dusty-nv, 2021-01-05:
Can you try putting the `output = jetson.utils.videoSource(...)` line above the `input = jetson.utils.videoSource(...)` line?

neilyoung, 2021-01-05:
@dusty-nv Interesting plot twist :) This works indeed. What is the voodoo? :)

Glad that it works, although I don't know why...

dusty-nv, 2021-01-05:
I'm not sure why either but its some weird interaction with the RTSP plugin that GStreamer uses and OpenGL.

neilyoung, 2021-01-05:
Hmm (...shakes his head doubtfully...)... But you did know about? Is there more of this kind? :)

dusty-nv, 2021-01-06:
I haven't been able to reproduce it

neilyoung, 2021-01-06:
Hmm. Anyway, this was very present here and did survive several reboots... Strange. 
But it is solved.

closed 2021-01-06 by neilyoung