dusty-nv/jetson-inference #886: jetson.utils.cudaOverlay transparent PNG support?

brewtangclan, 2021-01-08:
I am doing object detection on an RTSP stream of a fixed-mount camera, and want to be able to ignore certain areas within the camera's view. I'm looking to achieve this using a PNG of the "masked" area (black= ignored area, transparent = detection area). I have generated such an image, but using jetson.utils.cudaOverlay, the camera image is not visible behind the mask image - the masked area is black, but the rest of the image is white (jpg) or transparent (png).

test.py:
```
import jetson.utils
import jetson.inference

camera_image_path = 'camera_image.jpg'
mask_image_path = 'mask.png'
masked_image_path = 'masked_camera_image.jpg'

cam_img = jetson.utils.loadImage( camera_image_path, 'rgba8' )
mask_img = jetson.utils.loadImage( mask_image_path, 'rgba8' )

masked_img = jetson.utils.cudaAllocMapped( width=cam_img.width, height=cam_img.height, format=cam_img.format )
jetson.utils.cudaOverlay( cam_img, masked_img, 0, 0 )
jetson.utils.cudaOverlay( mask_img, masked_img, 0, 0 )
jetson.utils.cudaDeviceSynchronize()

jetson.utils.saveImage( masked_image_path, masked_img )
```
Script output:
```
[image] loaded 'camera_image.jpg'  (3840x2160, 3 channels)
[image] loaded 'mask.png'  (3840x2160, 3 channels)
[image] saved 'masked_camera_image.jpg'  (3840x2160, 3 channels)
```
Expected result: masked_camera_image.jpg, with masked area from mask.png "layered" superimposed over it.
Actual result: camera image not visible whatsoever in masked_camera_image.jpg (same if saving as .png, but the unmasked area is transparent instead of white).

If this is by design, I'm open to workarounds, though performance is paramount here, as I am looking to do this in as near to real-time as possible.

dusty-nv, 2021-01-08:
Hi @brewtangclan, the cudaOverlay function doesn't blend images - originally I made it to simple compost two images into a single image that weren't overlapping.   You can see in the code here that it simply copies the input pixel to the output:

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/cuda/cudaOverlay.cu#L39

However you could modify this to read both the input and output pixels, and then perform alpha blending between the input and output (and then write back out the output).   There is an alpha blending function here:

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/cuda/cudaAlphaBlend.cuh

brewtangclan, 2021-01-08:
Fair enough! Thanks for the quick response.

closed 2021-01-08 by brewtangclan