dusty-nv/jetson-inference #888: How to use a custom model in your coded image recognition program?

johnSheer, 2021-01-12:
Hi, I am new to this whole field, as my code is about to give away. I am having trouble using a image recognition model that I retrained with my program I am trying to code. Any help would be very much appreciated. 

import jetson.inference
import jetson.utils
import sys

import argparse

parser = argparse.ArgumentParser()
parser.add_argument("--model", type=str, default="~/Desktop/helloAiWorld/jetson-inference/python/training/classification/models/test1/resnet18.onnx", help="Help me.")
parser.add_argument("--labels", type=str, help="none")
parser.add_argument("filename", type=str, help="filename of the image to process")
parser.add_argument("--input-blob", type=str)
parser.add_argument("--output-blob", type=str)
#parser.add_argument("--batch_size", type=int, help="")
args = parser.parse_args()

#load Image
img = jetson.utils.loadImage(args.filename)

#load model
model = jetson.inference.imageNet(args.model, sys.argv )

#classify the image
class_idx, confidence = model.Classify(img)

#get class name
class_name = model.GetClassDesc(class_idx)

#print out the results
print('Classified image as {:s} (Class ID {:d}, confidence {:f})'.format(class_name,class_idx,confidence))

dusty-nv, 2021-01-12:
Hi @johnSheer, are you able to first load it with the `imagenet` program like shown here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#processing-images-with-tensorrt

Does it work if you change this line of your code?

``` python
model = jetson.inference.imageNet(argv=sys.argv)
```

If not, what is the console log?

johnSheer, 2021-01-12:
Dusty, thank you very much. Your change fixed it. I really appreciate the levels of abstraction you have established for beginners and I really look forward to better understanding the tools you have made available. It is simply brilliant what you have done.

closed 2021-01-12 by johnSheer

dusty-nv, 2021-01-12:
@johnSheer glad that fixed it for you and that you are finding the project useful!