dusty-nv/jetson-inference #889: Python custom argument behaviour

philnguyenresson, 2021-01-13:
Hi, I'm trying the DetectNet example code. If I run the default code, it'll load the model like so:

> detectNet -- loading detection network model from:
>           -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
>           -- input_blob   'Input'
>           -- output_blob  'NMS'
>           -- output_count 'NMS_1'
>           -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
>           -- threshold    0.500000
>           -- batch_size   1
> 
Is there documentation about each of these arguments? The documentation only mentions the parameters below:
If I explicitly add the model path to sys.argv `--model=networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff `(which is the same as the default model path), then I get

> detectNet -- loading detection network model from:
>           -- prototxt     NULL
>           -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
>           -- input_blob   'data'
>           -- output_cvg   'coverage'
>           -- output_bbox  'bboxes'
>           -- mean_pixel   0.000000
>           -- mean_binary  NULL
>           -- class_labels NULL
>           -- threshold    0.500000
>           -- batch_size   1

These are the parameters which are documented here (https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#detectNet). 
I have an SSD mobile net v2 model (stored in .onnx) that I've trained in tensorflow  but seperately from the jetson-inference library (so likely a slightly different network). I do know the names of the input and output layers and what they expect and output if I were to run them in tensorflow.  What am I supposed to be passing as sys.argv?

dusty-nv, 2021-01-21:
Hi @philnguyenresson , custom UFF models aren't able to be loaded from the command line, because they require additional parameters which aren't supported on the command line.

ONNX models can be though, with a similar syntax as shown here:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#processing-images-with-tensorrt

Note that the ONNX detection models are setup for the SSD-Mobilenet model trained with PyTorch from this Hello AI World tutorial though.  Meaning, the pre/post-processing for the model is what the PyTorch-based SSD-Mobilenet expects.  That may be different for your TensorFlow model, I'm not sure.   This is where the pre/post-processing code is:

* https://github.com/dusty-nv/jetson-inference/blob/5718df1495859808d5d076efea597b7e641bcb96/c/detectNet.cpp#L729
* https://github.com/dusty-nv/jetson-inference/blob/5718df1495859808d5d076efea597b7e641bcb96/c/detectNet.cpp#L813

Also, you should first check that your ONNX model can be loaded with TensorRT by using the `trtexec` tool found under `/usr/src/bin/tensorrt`

philnguyenresson, 2021-01-22:
Thanks @dusty-nv, it looks like my ONNX model fails the trtexec tool, so I'll have to fix that.

I did notice that pre/postprocessing is different. Is there a way to get the model to run without one of the preprocessors (like cudaTensorNormMeanRGB) being run on the input?

dusty-nv, 2021-01-22:
> Is there a way to get the model to run without one of the preprocessors (like cudaTensorNormMeanRGB) being run on the input?

Typically not, because the images/video come in as packed RGB (in HWC layout - Height*Width*Channels) with pixel values 0-255, but the models all typically use planar CHW layout (Channels*Height*Width) with normalized pixel values.  So if you want to get valid results, the same pre-processing needs to be applied as was done in PyTorch/TensorFlow.

philnguyenresson, 2021-01-22:
I see, it seems my model is quite a bit different from the ones in this repo. In my model the preprocessing is done on the as part of the tensorflow graph, but I now see that this is part of what is causing trtexec errors (UINT8 not supported). I will have to look into modifying the graph.
Is the planar CHW layout also a constraint of TRT, or is it just the way the models in the repo were designed?

dusty-nv, 2021-01-22:
> Is the planar CHW layout also a constraint of TRT, or is it just the way the models in the repo were designed?

TRT itself doesn't care, it CHW (NCHW) is just the most common input layout into DNN's

philnguyenresson, 2021-01-22:
I see, thanks for the help

closed 2021-01-22 by philnguyenresson