dusty-nv/jetson-inference #892: Jetson Nano - impossible to use three USB cams (seemingly)

neilyoung, 2021-01-17:
Hi,

I'm now banging my head to the wall for several days (and tested several different camera sets now, at least three...). It is for sure not a problem of this project, but probably of the Jetson at all.

1) I'm having three USB cams, each of which displaying these caps:

```
neil@jetson:~/jetson-inference/build/aarch64/bin$ gst-device-monitor-1.0 Video/Source
Probing devices...

Device found:

        name  : HD USB Camera
        class : Video/Source
        caps  : video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)6/1;
                video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)6/1;
                video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)9/1;
                video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)6/1;
                video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)20/1;
                video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
                video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
                image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
                image/jpeg, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
                image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1;
                image/jpeg, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
                image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1;
                image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)61612/513;
                image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)61612/513;
        properties:
                udev-probed = true
                device.bus_path = platform-70090000.xusb-usb-0:2.2:1.0
                sysfs.path = /sys/devices/70090000.xusb/usb1/1-2/1-2.2/1-2.2:1.0/video4linux/video0
                device.bus = usb
                device.subsystem = video4linux
                device.vendor.id = 32e4
                device.vendor.name = "HD\\x20USB\\x20Camera"
                device.product.id = 9230
                device.product.name = "HD\ USB\ Camera"
                device.serial = HD_USB_Camera_HD_USB_Camera
                device.capabilities = :capture:
                device.api = v4l2
                device.path = /dev/video0
                v4l2.device.driver = uvcvideo
                v4l2.device.card = "HD\ USB\ Camera"
                v4l2.device.bus_info = usb-70090000.xusb-2.2
                v4l2.device.version = 264588 (0x0004098c)
                v4l2.device.capabilities = 2216689665 (0x84200001)
                v4l2.device.device_caps = 69206017 (0x04200001)
        gst-launch-1.0 v4l2src ! ...

```

2) I'm not able to open three of these in a row using this simple Python script.

```
import cv2

streams = []
indexes = [0, 1, 2]

for index in indexes:
    stream = cv2.VideoCapture(index)
    stream.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter_fourcc(*"MJPG"))
    stream.set(cv2.CAP_PROP_FRAME_WIDTH, 640)
    stream.set(cv2.CAP_PROP_FRAME_HEIGHT, 480)
    streams.append(stream)

while True:
    i = 0
    for index in indexes:
        _, frame = streams[i].read()
        frame = cv2.resize(frame, (320, 240))
        cv2.imshow("Display "+str(index), frame)
       
        i = i + 1
    
    if cv2.waitKey(1) & 0xFF == ord('q'):
          break

```

3) The error is **always**:

```
neil@jetson:~/jetson-inference/build/aarch64/bin$ python3 test.py
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1184) setProperty OpenCV | GStreamer warning: GStreamer: unhandled property
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1184) setProperty OpenCV | GStreamer warning: GStreamer: unhandled property
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src2 reported: Failed to allocate required memory.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
VIDIOC_STREAMON: No space left on device
Traceback (most recent call last):
  File "test.py", line 22, in <module>
    frame = cv2.resize(frame, (320, 240))
cv2.error: OpenCV(4.1.1) /home/nvidia/host/build_opencv/nv_opencv/modules/imgproc/src/resize.cpp:3720: error: (-215:Assertion failed) !ssize.empty() in function 'resize'

```

4) **Now it comes: The same script runs absolutely fine on a Raspberry PI 4B... **

5) What are the differences? RPI has kernel 5.x, Jetson 4.9. RPI is using OpenCV 3.4.11, Jetson OpenCV 4.1.1. RPI is having 2 USB2 and 2 USB3 plugs, Jetson is having 4 USB3 plugs.

I have no idea anymore. I already tried to apply a patch @dusty-nv provided the link for https://github.com/dusty-nv/jetson-inference/issues/882#issuecomment-758807366 - to no avail. 

Anybody having an idea what to try else? Since I now already having spent too much time in solving this stupid capture issue on the Jetson I'm very close to give up with this platform at all.

neilyoung, 2021-01-18:
I was now trying to use a different pipeline, the result is the same. Even with 320 x 240 (!!)

```
gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=YUY2,width=320,height=240,framerate=30/1 ! nvvidconv ! 'video/x-raw(memory:NVMM),format=NV12' ! nvvidconv ! nveglglessink
````

The third opening leads to “Failed to allocate memory”:

```
neil@jetson:~$ gst-launch-1.0 v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! nvvidconv ! 'video/x-raw(memory:NVMM),format=NV12' ! nvvidconv ! nveglglessink
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate required memory.
Additional debug info:
gstv4l2src.c(658): gst_v4l2src_decide_allocation (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Buffer pool activation failed
Execution ended after 0:00:01.014041254
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

````

neilyoung, 2021-01-18:
Hello?

dusty-nv, 2021-01-18:
Sorry, it is a holiday here today - looks like you have a forum topic open for this issue (https://forums.developer.nvidia.com/t/three-cameras-on-jetson-nano/166052)

Since this issue doesn't seem because of jetson-inference, recommend to await our engineer's reply there.  Wish you the best of luck with your cameras!

neilyoung, 2021-01-19:
Yes, but the responsiveness there is pure compared to yours :)

Since it seems to be a common problem for the Jetson I think I will give up messing with it.

Thanks for your help

neilyoung, 2021-01-20:
Not getting any useful response there. Back on this issue list with this.

iliesaya, 2021-02-04:
I believe you can't stream 3 usb webcam because the USB bandwidth is exceeded. 2 USB and 1 MIPI camera should work.

neilyoung, 2021-02-04:
It works. Just a matter of compression and resolution. 640x480@30fps MJPEG works

closed 2021-02-04 by neilyoung