dusty-nv/jetson-inference #893: How can i get my rtsp channel 2 with jetson

ruiming46zrm, 2021-01-18:
Hi , thanks for your project.
    Now I want to pull the rtsp://<...>:port/Streaming/Channels/2 , it failed.  Channels/1 in ok
    How can i pull channel 2 ?
   (channel2 is lower quality in stream and faster during stream decode)

githublexluther, 2021-01-20:
can't even run it with one!

dusty-nv, 2021-01-21:
Can you post the terminal log from the console that you get?

If it says it can't detect the codec, you can use `--input-codec` to specify it manually.

ruiming46zrm, 2021-01-21:
> Can you post the terminal log from the console that you get?
> 
> If it says it can't detect the codec, you can use `--input-codec` to specify it manually.

like this:
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin:lettin20@192.168.2.220
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/2
Traceback (most recent call last):
  File "zrm_demo0.py", line 38, in <module>
    vs = VideoCaptureAsync('rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/2')
  File "zrm_demo0.py", line 9, in __init__
    self.cap = jetson.utils.videoSource(src)
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2021-01-21:
What happens if you set the `--input-codec=h264` option (or change to other codec if your stream is using a different codec)

``` bash
--input-codec=CODEC    RTP requires the codec to be set, one of these:
                             * h264, h265
                             * vp8, vp9
                             * mpeg2, mpeg4
                             * mjpeg
```

ruiming46zrm, 2021-01-22:
i tried every one , all is failed to create decoder for channel2, but opencv and deepstream is all ok.

ruiming46zrm, 2021-01-22:
and today  I meet a So strange issue:
```
import jetson.utils
import jetson.inference
import argparse
import sys

inp = jetson.utils.videoSource('rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/1')
outp = jetson.utils.videoOutput()

while True:
    image = inp.Capture()
    outp.Render(image)
```
when above codes runs , no output come out , and output video comes out  if I delete "import jetson.inferense"  like following:
```
import jetson.utils
import argparse
import sys

inp = jetson.utils.videoSource('rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/1')
outp = jetson.utils.videoOutput()

while True:
    image = inp.Capture()
    outp.Render(image)
```

ruiming46zrm, 2021-01-22:
this strange issue solved by change the inp oup order:
```
import jetson.utils
import jetson.inference
import argparse
import sys

outp = jetson.utils.videoOutput()
inp = jetson.utils.videoSource('rtsp://admin:lettin20@192.168.2.220:554/Streaming/Channels/1')

while True:
    image = inp.Capture()
    outp.Render(image)
```

but I'm currious about the reason

ruiming46zrm, 2021-02-26:
@dusty-nv   hi , dusty,

for a period of time , I stiil can not find the way to pull my sub stream, can you tell me the reason or any method?

closed 2023-04-27 by dusty-nv