dusty-nv/jetson-inference #896: Custom model architecture

RGring, 2021-01-20:
Hi @dusty-nv,
Thanks for the nice repository! 

Is it possible to load any model architecture as long as it is converted to the onxx-format?
If not, could you give me some pointers, what must be adapted if I want to get it working?

Thanks in advance for your time!

dusty-nv, 2021-01-20:
Hi @RGring , there is also pre/post-processing code that goes along with the models.   You can find this code in `imageNet.cpp`, `detectNet.cpp`, `segNet.cpp` (found under the `jetson-inference/c/` directory).

For example, with pre-processing there is mean pixel subtraction, standard deviation/normalization, and NCHW format conversion applied.   A lot of times this is similar for PyTorch models - for example, here is pre-processing for ONNX ssd-mobilenet model trained with PyTorch:

https://github.com/dusty-nv/jetson-inference/blob/5718df1495859808d5d076efea597b7e641bcb96/c/detectNet.cpp#L729

TensorFlow and Caffe typically do their pre-processing slightly differently so different pre-processing functions are called.

Regarding the post-processing, that can vary more based on the model/network.  For example with detection, the bounding boxes and confidences need interpreted.  Different detection networks output their detection in different structures.  Here is the one for ssd-mobilenet:

https://github.com/dusty-nv/jetson-inference/blob/5718df1495859808d5d076efea597b7e641bcb96/c/detectNet.cpp#L815

Also, before spending much time supporting a new model, you want to check that the ONNX can be loaded with TensorRT using the `trtexec` tool (found under `/usr/src/tensorrt/bin`)

t-T-s, 2021-01-25:
Hi @dusty-nv,
First of all this repository is really awesome. I can't wait to use it for all my upcoming projects.

### Prerequisites:
**Jetson:** _[jetson nano (JetPack 4.4.1)]_

I tried to run a model I trained using tensorflow. Then I converted the model to ONNX. It went smoothly. I used [tf2onnx](https://github.com/onnx/tensorflow-onnx). Here's the command I used for that. I had to use opset 11 since my model `ssd_mobilenet_v2_fpnlite_320x320_coco17_tpu-8` is from tensorflow model zoo and it had `NonMaxSupression` (which contributes to my problem) layer in it.

```
$ python -m tf2onnx.convert --saved-model "<saved-model-base>\saved" --output "<output-model-base>\ssd-mobilenet.onnx" --opset 11
```
Then came the problem of unsupported Dtype UINT8. I was also able to resolve this problem. I used [onnx-graphsurgeon](https://github.com/NVIDIA/TensorRT/tree/master/tools/onnx-graphsurgeon). Here's the code for that.
```
import onnx_graphsurgeon as gs
import onnx
import numpy as np

graph = gs.import_onnx(onnx.load("model/onnxmodel/ssd-mobilenet-v2-fpnlite.onnx"))
for inp in graph.inputs:
    inp.dtype = np.float32

onnx.save(gs.export_onnx(graph), "model/onnxmodel/ssd-mobilenet-v2-fpnlite-up.onnx")
```

### Problem:
I also had to replace the NMS layer.  This is where I'm stuck at. Here is the code I used to do that. 
```
import onnx_graphsurgeon as gs
import onnx
import numpy as np

input_model_path = "model/onnxmodel/ssd-mobilenet-v2-fpnlite_updated.onnx"
output_model_path = "model/onnxmodel/ssd-mobilenet-v2-fpnlite_updated_nms.onnx"

@gs.Graph.register()
def trt_batched_nms(self, boxes_input, scores_input, nms_output,
                    share_location, num_classes):

    boxes_input.outputs.clear()
    scores_input.outputs.clear()
    nms_output.inputs.clear()

    attrs = {
        "shareLocation": share_location,
        "numClasses": num_classes,
        "backgroundLabelId": -1,
        "topK": 1024,
        "keepTopK": 100,
        "scoreThreshold": 0.0001,
        "iouThreshold": 0.6,
        "isNormalized": True,
        "clipBoxes": True
    }
    return self.layer(op="BatchedNMS_TRT", attrs=attrs,
                      inputs=[boxes_input, scores_input],
                      outputs=[nms_output])

graph = gs.import_onnx(onnx.load(input_model_path))

print(graph.inputs[0].shape)

tmap = graph.tensors()

graph.trt_batched_nms(tmap["Unsqueeze__695:0"],
                      tmap["Unsqueeze__798:0"],
                      tmap["NonMaxSuppression__800:0"],
                      share_location=False,
                      num_classes=6)

graph.trt_batched_nms(tmap["Unsqueeze__695:0"],
[... 60 lines cut ...]
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: largest_int_val__984
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: StatefulPartitionedCall/Postprocessor/BatchMultiClassNonMaxSuppression/PadOrClipBoxList/zeros_10:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: StatefulPartitionedCall/Postprocessor/BatchMultiClassNonMaxSuppression/PadOrClipBoxList/zeros:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: StatefulPartitionedCall/Postprocessor/BatchMultiClassNonMaxSuppression/PadOrClipBoxList/sub_17/x:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: StatefulPartitionedCall/Postprocessor/BatchMultiClassNonMaxSuppression/PadOrClipBoxList/sub_3/x:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: const_fold_opt__1161
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:90: Importing initializer: const_fold_opt__1160
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:103: Parsing node: StatefulPartitionedCall/Cast [Cast]
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:119: Searching for input: input_tensor:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:125: StatefulPartitionedCall/Cast [Cast] inputs: [input_tensor:0 -> (1, -1, -1, 3)], 
[01/25/2021-07:01:53] [V] [TRT] builtin_op_importers.cpp:320: Casting to type: float32
[01/25/2021-07:01:53] [V] [TRT] ImporterContext.hpp:141: Registering layer: StatefulPartitionedCall/Cast for ONNX node: StatefulPartitionedCall/Cast
[01/25/2021-07:01:53] [V] [TRT] ImporterContext.hpp:116: Registering tensor: StatefulPartitionedCall/Cast:0 for ONNX tensor: StatefulPartitionedCall/Cast:0
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:179: StatefulPartitionedCall/Cast [Cast] outputs: [StatefulPartitionedCall/Cast:0 -> (1, -1, -1, 3)], 
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:103: Parsing node: onnx_graphsurgeon_node_3 [BatchedNMS_TRT]
[01/25/2021-07:01:53] [V] [TRT] ModelImporter.cpp:125: onnx_graphsurgeon_node_3 [BatchedNMS_TRT] inputs: 
[01/25/2021-07:01:53] [I] [TRT] ModelImporter.cpp:135: No importer registered for op: BatchedNMS_TRT. Attempting to import as plugin.
[01/25/2021-07:01:53] [I] [TRT] builtin_op_importers.cpp:3659: Searching for plugin: BatchedNMS_TRT, plugin_version: 1, plugin_namespace: 
[01/25/2021-07:01:53] [I] [TRT] builtin_op_importers.cpp:3676: Successfully created plugin: BatchedNMS_TRT
Segmentation fault (core dumped)
```
Full log text: [log.txt](https://github.com/dusty-nv/jetson-inference/files/5863795/log.txt)

Can you please help me with this? It's such an optimized way to deploy the model and it would also be a great learning opportunity.
Thank you in advance.

dusty-nv, 2021-01-25:
Hi @t-T-s, unfortunately I'm not very familiar with converting detection models from TensorFlow to TensorRT.   This is the tool I used previously:  https://github.com/AastaNV/TRT_object_detection

However these days I train SSD-Mobilenet in PyTorch (as shown in this repo), then convert it to ONNX and load it with TensorRT.  That seems to work more smoothly.   Sorry to not be of more help with TensorFlow.

t-T-s, 2021-01-26:
Oh dat's alright @dusty-nv . I was also thinking about either going with TF-TRT or PyTorch as you mentioned. Thank you for the repo. It's amazing. I'll try to use the AastaNV's tool. Happy training.

VeeranjaneyuluToka, 2021-04-26:
@t-T-s , are you abel to convert your tensorflow ssd-mobilenet-v2-fpnlite model to trt engine?

t-T-s, 2021-04-27:
@VeeranjaneyuluToka  No. I had to use TF-TRT and move on. Still looking for a solution though.

closed 2023-04-27 by dusty-nv