dusty-nv/jetson-inference #897: Retraining Custom Object Detection Model

naimulhq, 2021-01-21:
I have been able to create my own custom object detection that is designed to detect various objects and converted the model to .onnx so that I can use my raspberry pi camera. After working on detecting various objects, I am trying to add new objects for the model to recognize and know that I would need to train another model. I was wondering how I would go about retraining the existing model I have to recognize new objects? I heard about transfer learning but do not fully understand how to set it up.

dusty-nv, 2021-01-25:
Hi @naimulhq, have you added the new object(s) to your previous dataset?   All the objects you want to recognize would need to be in the same dataset.   Otherwise, it will only train the model on the new objects only.

You should be able to use the [`--pretrained`](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/train_ssd.py#L51) argument to train_ssd.py and point it to your previous model.  Or if you have trouble, just train it again from the beginning on your completed dataset (the dataset with both the old objects and new objects)

naimulhq, 2021-01-25:
Hi @dusty-nv , No I haven't. Just to be clear, let say as an example I have trained a model which is able to detect cars but now, I would like to make that same model be able to detect an airplane. I would have to add the images and annotations for the airplane in the existing dataset for cars?

dusty-nv, 2021-01-25:
> I would have to add the images and annotations for the airplane in the existing dataset for cars?

Yes that is correct, otherwise the model would only detect cars.

naimulhq, 2021-01-25:
@dusty-nv Lastly, when I use the --pretrained argument, do I point it to the .onnx file or .pth?

dusty-nv, 2021-01-25:
> Lastly, when I use the --pretrained argument, do I point it to the .onnx file or .pth?

The .pth - the last pth that was output

naimulhq, 2021-01-25:
@dusty-nv Thank you very much!

closed 2021-01-25 by naimulhq