dusty-nv/jetson-inference #898: FYI: Problem downloading images to run object detection demo

mcclumpherty, 2021-01-21:
I've just bought myself a shiny 4Gb Jetson Nano and am so far very impressed with it.  I thought I would run through the SSD MobileNet retraining demo on it.

However when I first tried it I got the following error when running it via the docker container:

```
root@stirling:/jetson-inference/python/training/detection/ssd# python3 ./open_images_downloader.py --class-names "Apple,Orange,Banana,Strawberry,Grape,Pear,Pineapple,Watermelon" --data=data/fruit
2021-01-21 16:47:50 - Download https://storage.googleapis.com/openimages/2018_04/class-descriptions-boxable.csv.
2021-01-21 16:47:50 - Requested 8 classes, found 8 classes
2021-01-21 16:47:50 - Download https://storage.googleapis.com/openimages/2018_04/train/train-annotations-bbox.csv.
2021-01-21 16:48:59 - Read annotation file data/fruit/train-annotations-bbox.csv
Traceback (most recent call last):
  File "./open_images_downloader.py", line 144, in <module>
    annotation_set = pd.read_csv(annotation_file)
  File "/usr/local/lib/python3.6/dist-packages/pandas/io/parsers.py", line 686, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "/usr/local/lib/python3.6/dist-packages/pandas/io/parsers.py", line 458, in _read
    data = parser.read(nrows)
  File "/usr/local/lib/python3.6/dist-packages/pandas/io/parsers.py", line 1211, in read
    df = DataFrame(col_dict, columns=columns, index=index)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/frame.py", line 468, in __init__
    mgr = init_dict(data, index, columns, dtype=dtype)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/construction.py", line 283, in init_dict
    return arrays_to_mgr(arrays, data_names, index, columns, dtype=dtype)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/construction.py", line 93, in arrays_to_mgr
    return create_block_manager_from_arrays(arrays, arr_names, axes)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/managers.py", line 1683, in create_block_manager_from_arrays
    blocks = form_blocks(arrays, names, axes)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/managers.py", line 1753, in form_blocks
    int_blocks = _multi_blockify(items_dict["IntBlock"])
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/managers.py", line 1835, in _multi_blockify
    values, placement = _stack_arrays(list(tup_block), dtype)
  File "/usr/local/lib/python3.6/dist-packages/pandas/core/internals/managers.py", line 1863, in _stack_arrays
    stacked = np.empty(shape, dtype=dtype)
MemoryError: Unable to allocate 669. MiB for an array with shape (6, 14610229) and data type int64
```
In hope, I ran it a second time and it worked.  Not sure why this would be the case.  On the second run I monitored it with jtop and it certainly used all the available RAM and swap space, but at least it didn't fall over and successfully downloaded all the images.

dusty-nv, 2021-01-21:
Hi @mcclumpherty , have you tried mounting additional SWAP space like here: 

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap

Also, you might want to disable ZRAM (since it consumes physical memory) by running:

``` bash
sudo systemctl disable nvzramconfig
```

mcclumpherty, 2021-01-24:
Hi Dusty,

Many thanks for getting back to me so quickly !  

Yes, in the end I set up a 10Gb swap space on the SSD and that got it to work.
I was able to run the training then using:

`python3 train_ssd.py --data=data/fruit --models-dir=models/fruit --batch-size=2 --workers=1 --epochs=5`

It took about 2 hours 10  minutes to run through 5 epochs.  What was curious was that for most of the time the memory usage sat around 6Gb (the full 4Gb of RAM + 2Gb of swap with no other processes running), but twice during the 5 epochs, the memory usage went mad and used up to 9Gb of swap space.  At that point, the training seemed to hang for about 8 minutes while it sorted itself out and then continued.

At the end of the 5 epochs, I continued to try to reproduce your demo and did the ONNX conversion and ran detectnet on the fruit.  However, the results were nowhere near as good as yours where objects were only detected in 4 out of the 21 test images.

Do you have any idea why this might be the case?

closed 2023-04-27 by dusty-nv