dusty-nv/jetson-inference #899: cmake error

chenbokaix250, 2021-01-22:
When i start `cmake../`
error is **Python 3.7 wasn't found**

so i install python3.7 
command is `sudo apt-get install libpython3.7-dev`

next i start  `camke ../`
error is NumPy not found

```
-- detecting Python 3.7...
-- found Python version:  3.7 (3.7.5)
-- found Python include:  /usr/include/python3.7m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.7m.so
-- CMake module path:  /home/cold/jetson-inference/utils/cuda;/home/cold/jetson-inference/python/bindings;/home/cold/jetson-inference/python/bindings/../../utils/python/bindings
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'numpy'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 24, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'

Original exception was:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'numpy'
-- NumPy not found
-- Copying /home/cold/jetson-inference/python/examples/detectnet.py
-- Copying /home/cold/jetson-inference/python/examples/imagenet.py
-- Copying /home/cold/jetson-inference/python/examples/my-detection.py
-- Copying /home/cold/jetson-inference/python/examples/my-recognition.py
-- Copying /home/cold/jetson-inference/python/examples/segnet.py
-- Copying /home/cold/jetson-inference/python/examples/segnet_utils.py
-- Copying examples/imagenet.py -> imagenet-console.py
-- Copying examples/imagenet.py -> imagenet-camera.py
-- Copying examples/detectnet.py -> detectnet-console.py
-- Copying examples/detectnet.py -> detectnet-camera.py
-- Copying examples/segnet.py -> segnet-console.py
-- Copying examples/segnet.py -> segnet-camera.py
-- Configuring incomplete, errors occurred!
See also "/home/cold/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/cold/jetson-inference/build/CMakeFiles/CMakeError.log".

```

but in Xavier I can't find the suitable numpy version

how can I solve this problem

dusty-nv, 2021-01-22:
That shouldn't be an error that Python 3.7 is not found, Python 3.7 isn't required.  Python 3.6 should work.

Can you install numpy into Python 3.7 with 3.7's pip?  Or if you don't actually want 3.7, uninstall it

chenbokaix250, 2021-01-23:
> That shouldn't be an error that Python 3.7 is not found, Python 3.7 isn't required. Python 3.6 should work. Can you install numpy into Python 3.7 with 3.7's pip? Or if you don't actually want 3.7, [...]

I can't install numpy for python3.7 with pip

thanks for your advice！

when I uninstall python3.7  I 

I found the error because not install Doxygen 

when I installed ，everythink is oK

thanks！

closed 2021-01-23 by chenbokaix250

AndreeaVM, 2021-04-19:
Hi! I have the same error. 
I installed python3.7, also the doxygen, but the error is still there. (I tried on Jetson NX and Nano)
Do you have any idea what should I do?

dusty-nv, 2021-04-19:
For Python 3.7, you will need to install numpy via pip3.7.  So first install pip for python3.7, and then install numpy from it.