dusty-nv/jetson-inference #900: How can I hide the gl title

lzcchl, 2021-01-22:
[image]
I want show frame with fullscreen, but I can't find a function to hide the gl window title, can you help me?

lzcchl, 2021-01-25:
Emm. I find an answer at "https://github.com/dusty-nv/jetson-inference/issues/801", but when I try it, it's also a title on the window,  is there a function to hide title?

dusty-nv, 2021-01-25:
Hi @lzcchl , there isn't a built-in function, but you could try the recommendation from this post:

https://forums.developer.nvidia.com/t/display-in-jetson-inference-without-title-bar-and-image-blending/165214/2?u=dusty_nv

flow-dev, 2021-02-17:
I was having trouble with full screen output, 
so it was very helpful!

I changed the code like this and it worked.

https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/display/glDisplay.cpp#L322

```
	// Add FullScreen output
	Atom wm_state   = XInternAtom (mDisplayX, "_NET_WM_STATE", true );
	Atom wm_fullscreen = XInternAtom (mDisplayX, "_NET_WM_STATE_FULLSCREEN", true );
	XChangeProperty(mDisplayX, win, wm_state, XA_ATOM, 32, PropModeReplace, (unsigned char *)&wm_fullscreen, 1);
```

lzcchl, 2021-02-19:
@flow-dev this is good!

closed 2021-02-19 by lzcchl