dusty-nv/jetson-inference #901: Jetson-inference container comaptiblity with Jetpack 4.5

mmatesic01, 2021-01-22:
Hi,
When can we expect jetson-inference container instalation compatibility with Jetpack 4.5?
Thanks.

dusty-nv, 2021-01-22:
Hi @mmatesic01 , last night I pushed the `jetson-inference:r32.5.0` container tag for JetPack 4.5 to dockerhub:

https://hub.docker.com/repository/docker/dustynv/jetson-inference/tags

The docker scripts in the repo were also updated with commit https://github.com/dusty-nv/jetson-inference/commit/2e5fdafcbb706d55102dc3c2bd93885a76a4a830
So if you pull the latest from master, it should work on JetPack 4.5.

mmatesic01, 2021-01-22:
I tried it but I get this:

/jetson-containers$ ./scripts/docker_build_ml.sh all
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.5.0
l4t-base image:  nvcr.io/nvidian/nvidia-l4t-base:r32.5.0
building PyTorch torch-1.6.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.7.0 (pillow), torchaudio v0.6.0
Building l4t-pytorch:r32.5.0-pth1.6-py3 container...
[sudo] password for marko: 
Sending build context to Docker daemon  215.6kB
Step 1/22 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.4
Step 2/22 : FROM ${BASE_IMAGE}
unauthorized: authentication required

dusty-nv, 2021-01-22:
``` bash
$ cd /path/to/your/jetson-inference
$ git pull origin master
$ docker/run.sh
```

mmatesic01, 2021-01-22:
Excellent.
Thank you.

closed 2021-01-22 by mmatesic01