dusty-nv/jetson-inference #902: Detection: One of three displays is black...

neilyoung, 2021-01-23:
Hi Dusty,

You know, I was fighting like hell for the Jetson (in comparison to an RPI/Coral). I would like a Jetson based solution more, since it is more compact after all. But as of now, the RPI/Coral tuple achieves double the frame rate with three cameras.

It took me two weeks to find a set of cameras, which worked more or less w/o complains on the Jetson (while none of the previous sets had a problem with the RPI). It was always this "memory allocation" thing, which basically was just a USB bus congestion on the Jetson dev kit.

But finally I found a brand, from which I was able to capture from three cams MJPEG simultaneously. H.264 also works kind of, but the latency is higher.

So now my question: Would you see a reason, why the **threaded approach** (which is 1 fps faster than the other and would be kind of future prove for me) does reproducible always show one black preview window out of three, which nevertheless counts up the frames in the headline? The same does not happen with the non-threaded approach, there I have all the windows showing the images.

I'm calling both variants like so:

```
python3 my-detection.py --input-codec=mjpeg --input-width=640 --input-height=480
```

I admit I'm a bit tired after two weeks fighting the Jetson while seeing the RPI/Coral running good. So this is a last attempt with this platform and you will finally get rid of me (promised), if I cannot solve it.

Threaded approach:

```python
#!/usr/bin/python3
import jetson.inference
import jetson.utils
import sys
import time
from threading import Thread, Lock

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5, argv=sys.argv)

cameras = [
    jetson.utils.videoSource("/dev/video0", argv=sys.argv),     
    jetson.utils.videoSource("/dev/video2", argv=sys.argv),
    jetson.utils.videoSource("/dev/video4", argv=sys.argv)  
]

displays = [
    jetson.utils.videoOutput("display://0", argv=sys.argv),  
    jetson.utils.videoOutput("display://0", argv=sys.argv),  
    jetson.utils.videoOutput("display://0", argv=sys.argv),  
]

inference_lock = Lock()
inferences = []
detections = []

class Inference:
    def __init__(self, index=0):
        self.index = index
    
    def start(self):
        Thread(target=self.infer, args=()).start()

    def infer(self):
        while True:
            frame = cameras[self.index].Capture()
            
            inference_lock.acquire()
            detections[self.index] = net.Detect(frame)
            inference_lock.release()
            
            displays[self.index].Render(frame)
            displays[self.index].SetStatus("Object Detection | Network {:.0f} FPS".format(displays[self.index].GetFrameRate()))

for index in range(len(cameras)):
    inferences.append(Inference(index))
    detections.append(None)
    inferences[index].start()

while displays[0].IsStreaming():
    pass
```

Non-threaded approach:

```python
#!/usr/bin/python3

import jetson.inference
import jetson.utils
import sys
import time

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5, argv=sys.argv)
cameras = [
    jetson.utils.videoSource("/dev/video0", argv=sys.argv),     
    jetson.utils.videoSource("/dev/video2", argv=sys.argv),
    jetson.utils.videoSource("/dev/video4", argv=sys.argv)  
]

displays = [
    jetson.utils.videoOutput("display://0"),
    jetson.utils.videoOutput("display://0"),	
    jetson.utils.videoOutput("display://0")
]

detections = []
images = []

for index in range(len(cameras)):
    detections.append(None)
    images.append(None)

while displays[0].IsStreaming():
    for index in range(len(cameras)):
        images[index] = cameras[index].Capture()
        detections[index] = net.Detect(images[index])
        displays[index].Render(images[index])
        displays[index].SetStatus("Object Detection | Network {:.0f} FPS".format(displays[index].GetFrameRate()))
```

Don't get me wrong please. I don't want you to fix my code. I'm sure the code is OK so far (under the assumption, that threading generally works with the lib). But I can't see an error anywhere, maybe I just need a hint where to look.

I also tried to play with `--output-width`and `--output-height` to no avail.

TIA

neilyoung, 2021-01-23:
BTW: The `inference_lock` doesn't seem to make sense at all. It also works w/o (magically...)

dusty-nv, 2021-01-25:
The videoSource is already threaded underneath so I'm not sure adding yet another thread to it would do all that much.  Also I haven't tried openGL multithreaded but it may not like more than one rendering thread at once - not sure.

You could try jetson.utils.cudaOverlay to make 1 big image out of the streams, to one window.  Maybe that is faster than 3 images.

DeepStream does better handling of several camera streams than jetson-inference does.

neilyoung, 2021-01-25:
Thanks for letting me know. Will check this out.

neilyoung, 2021-01-25:
Hmm. I must be missing something. I have now changed my threaded approach like so.

Now the entire full width image of 1920 x 480 is black :)

```
#!/usr/bin/python3

import jetson.inference
import jetson.utils
import sys
import time

from threading import Thread, Lock

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5, argv=sys.argv)

display = jetson.utils.videoOutput("display://0", argv=sys.argv)

cameras = [
    jetson.utils.videoSource("/dev/video0", argv=sys.argv),     
    jetson.utils.videoSource("/dev/video2", argv=sys.argv),
    jetson.utils.videoSource("/dev/video4", argv=sys.argv)  
]

inference_lock = Lock()
inferences = []
detections = []
counters = []
fps = []
images = []

class Inference:
    def __init__(self, index=0):
        self.index = index
    
    def start(self):
        Thread(target=self.infer, args=()).start()

    def infer(self):
        while True:
            images[self.index] = cameras[self.index].Capture()
            
            inference_lock.acquire()
            detections[self.index] = net.Detect(images[self.index])
            inference_lock.release()
            
            counters[self.index] = counters[self.index] + 1

            if all(image is not None for image in images):
                totalW = 0
                totalH = 0
                for image in images:
                    totalW = totalW + image.width
                    totalH = max(totalH, image.height)
                
                image_out = jetson.utils.cudaAllocMapped(width=totalW, height=totalH, format=images[0].format)
                
                print(image_out.width, image_out.height)
                
                # Stitch together
                x = 0
                for image in images:
                    jetson.utils.cudaOverlay(image, image_out, x, 0)
                    x = x + image.width 
                
                display.Render(image_out)

for index in range(len(cameras)):
    inferences.append(Inference(index))
    detections.append(None)
    images.append(None)
    counters.append(0)
    fps.append(0)
    inferences[index].start()

start_time = time.time()

while True:
    time.sleep(0.5)
    end_time = time.time()
    delta_t = end_time - start_time
    if delta_t >= 1.0:
        for i in range(len(inferences)):
            fps[i] = counters[i] / delta_t 
            counters[i] = 0
        
        for i in range(len(inferences)):
            print("FPS {0:d}: {1:.0f}".format(i, round(fps[i])))
            
        start_time = time.time()
```

neilyoung, 2021-01-25:
My first experiences with deep stream are not that superior as with your project. Would you mind to point me to a python sample, which comes close to my app?

neilyoung, 2021-01-25:
More specifically this is my first contact:

It doesn't work in either specification of the camera...

<pre>neil@jetson:~/deepstream_python_apps/apps/deepstream-test3$ python3 deepstream_test_3.py v42l:///dev/video1
Creating Pipeline 
 
Creating streamux 
 
Creating source_bin  0  
 
Creating source bin
source-bin-00
Creating Pgie 
 
Creating tiler 
 
Creating nvvidconv 
 
Creating nvosd 
 
Creating transform 
 
Creating EGLSink 

Adding elements to Pipeline 

Linking elements in the Pipeline 

Now playing...
1 :  v42l:///dev/video1
Starting pipeline 

Using winsys: x11 
ERROR: Deserialize engine failed because file path: /home/neil/deepstream_python_apps/apps/deepstream-test3/../../../../samples/models/Primary_Detector/resnet10.caffemodel_b1_gpu0_int8.engine open error
0:00:01.473882987 15166      0x628ef80 <font color="#FCE94F"><b>WARN   </b></font>              nvinfer gstnvinfer.cpp:616:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Warning from NvDsInferContextImpl::deserializeEngineAndBackend() &lt;nvdsinfer_context_impl.cpp:1690&gt; [UID = 1]: deserialize engine from file :/home/neil/deepstream_python_apps/apps/deepstream-test3/../../../../samples/models/Primary_Detector/resnet10.caffemodel_b1_gpu0_int8.engine failed
0:00:01.473953040 15166      0x628ef80 <font color="#FCE94F"><b>WARN   </b></font>              nvinfer gstnvinfer.cpp:616:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Warning from NvDsInferContextImpl::generateBackendContext() &lt;nvdsinfer_context_impl.cpp:1797&gt; [UID = 1]: deserialize backend context from engine from file :/home/neil/deepstream_python_apps/apps/deepstream-test3/../../../../samples/models/Primary_Detector/resnet10.caffemodel_b1_gpu0_int8.engine failed, try rebuild
0:00:01.473986530 15166      0x628ef80 <font color="#06989A">INFO   </font>              nvinfer gstnvinfer.cpp:619:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Info from NvDsInferContextImpl::buildModel() &lt;nvdsinfer_context_impl.cpp:1715&gt; [UID = 1]: Trying to create engine from model files
ERROR: Cannot access prototxt file &apos;/home/neil/deepstream_python_apps/apps/deepstream-test3/../../../../samples/models/Primary_Detector/resnet10.prototxt&apos;
ERROR: failed to build network since parsing model errors.
ERROR: failed to build network.
0:00:01.474237056 15166      0x628ef80 <font color="#EF2929"><b>ERROR  </b></font>              nvinfer gstnvinfer.cpp:613:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Error in NvDsInferContextImpl::buildModel() &lt;nvdsinfer_context_impl.cpp:1735&gt; [UID = 1]: build engine file failed
0:00:01.474273671 15166      0x628ef80 <font color="#EF2929"><b>ERROR  </b></font>              nvinfer gstnvinfer.cpp:613:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Error in NvDsInferContextImpl::generateBackendContext() &lt;nvdsinfer_context_impl.cpp:1821&gt; [UID = 1]: build backend context failed
0:00:01.474308516 15166      0x628ef80 <font color="#EF2929"><b>ERROR  </b></font>              nvinfer gstnvinfer.cpp:613:gst_nvinfer_logger:&lt;primary-inference&gt; NvDsInferContext[UID 1]: Error in NvDsInferContextImpl::initialize() &lt;nvdsinfer_context_impl.cpp:1148&gt; [UID = 1]: generate backend failed, check config file settings
0:00:01.474345339 15166      0x628ef80 <font color="#FCE94F"><b>WARN   </b></font>              nvinfer gstnvinfer.cpp:809:gst_nvinfer_start:&lt;primary-inference&gt; error: Failed to create NvDsInferContext instance
0:00:01.474368152 15166      0x628ef80 <font color="#FCE94F"><b>WARN   </b></font>              nvinfer gstnvinfer.cpp:809:gst_nvinfer_start:&lt;primary-inference&gt; error: Config file path: dstest3_pgie_config.txt, NvDsInfer Error: NVDSINFER_CONFIG_FAILED
Error: gst-resource-error-quark: Failed to create NvDsInferContext instance (1): /dvs/git/dirty/git-master_linux/deepstream/sdk/src/gst-plugins/gst-nvinfer/gstnvinfer.cpp(809): gst_nvinfer_start (): /GstPipeline:pipeline0/GstNvInfer:primary-inference:
Config file path: dstest3_pgie_config.txt, NvDsInfer Error: NVDSINFER_CONFIG_FAILED
Exiting app
</pre>

neilyoung, 2021-01-25:
Disregard my last post. I have overseen, that I need to clone the python repo under the source path of the DeepStream SDK root...

I was able to replay an MP4 file and do inference on it. But with none of my cams there is something useful to get. It just hangs :(

dusty-nv, 2021-01-25:
Hi @neilyoung , here are the Python samples for DeepStream, which it seems you have already found:   https://github.com/NVIDIA-AI-IOT/deepstream_python_apps

It looks like the error is unrelated to your camera:

``` bash
ERROR: Deserialize engine failed because file path: /home/neil/deepstream_python_apps/apps/deepstream-test3/../../../../samples/models/Primary_Detector/resnet10.caffemodel_b1_gpu0_int8.engine open error
```

INT8 is not for Nano, FP16 is.  So you may need to modify the config file or script.  If you continue to have issues with it, I recommend to post on the DeepStream forums or that GitHub.

neilyoung, 2021-01-25:
Yes, I was gracefully ignoring this. It took FP16 by default after failing INT8. It worked with my mp4 file. But now I'm trying to open /dev/video0 with the usb-cam-sample to no avail. There is a small window with a "question mark" to the left of the console (as if it could not open the display window) and that's it 

[image]

neilyoung, 2021-01-25:
@dusty-nv Have you seen my post regarding the completely black window with `cudaOverlay` now? Am I doing something wrong in my code?

neilyoung, 2021-01-25:
And maybe you can tell me, what the advantage of this DeepStream compared to your project is. It looks ugly and it doesn't work with my cameras (again)...

dusty-nv, 2021-01-25:
Typically DeepStream is adept at handling multiple camera streams simultaneously, sorry it is having problems with your cameras.  The DeepStream team on the forums may know more about it.  I will take a look at your other post about cudaOverlay.

neilyoung, 2021-01-25:
...but it definitely fails the target to encourage beginners. Unusually crappy...

dusty-nv, 2021-01-25:
I have been looking at your usage of cudaCrop(), and it looks similar to how I use it, so not sure why it is black, so I wonder if it is something to do with the threading or not.  It could be useful to try it in your original non-threaded app or just with some static test images first to narrow down the problem.  

Also, I typically do the memory allocation before entering the main camera loop, or on the first frame (since that is when you know the dimensions.   Above the main loop you can have `img_out=None` and then inside the loop check `if img_out is None:`.  Then if it is None, allocate it (instead of doing it every frame)

neilyoung, 2021-01-25:
Yes, it works well in the non-threaded version.

And yes, your suggestion is fine. Thanks for this.

One message from the other front (I know, it's not your party, but you are so helpful..): I got it running with _another_ camera (well, will this never stop...)  and the native app:

`deepstream-app -c /opt/nvidia/deepstream/deepstream-5.0/samples/configs/deepstream-app/source4_1080p_dec_infer-resnet_tracker_sgie_tiled_display_int8.txt
`

after having aligned the config to work with the camera.

I'm having a frame rate of about 30 fps, but the inference quality seems to be not that great: The drawn frame is mostly not framing me, or the lower edge is at the bottom of the display. The camera is an extreme fisheye - does that play a role?

I will now go and try with the python app and then I have to blame the camera again...

neilyoung, 2021-01-25:
I tell you what: With this camera it also works with the Python script. I could puke. It took me 2 weeks to find a camera set, which worked with three cams and your project. Now it turns out - nada. DeepStream again requires something different... Seriously...

dusty-nv, 2021-01-25:
Sorry about that, certainly it is frustrating.  It seems DeepStream uses a different gstreamer element for v4l2 capture:

https://docs.nvidia.com/metropolis/deepstream/5.0DP/plugin-manual/index.html#page/DeepStream%20Plugins%20Development%20Guide/deepstream_plugin_details.3.12.html

neilyoung, 2021-01-25:
No issue, Dusty. I'm learning with this. Just the time flies...

neilyoung, 2021-01-25:
But I still can't use even one of the cams, which worked fine with this project... Funny...

neilyoung, 2021-01-27:
Hi Dusty, I literally had the same problems with 3 cams on the Jetson as with your project. Right now it is working with three cams, all operated in 640x480 MJPEG capture input @ 30fps. The inference engine is making 20 fps per camera, which is tremendously more than what I have achieved with your project. But - the SDK is a real PITA, the code looks messy and compares in nothing with your easy to read, well documented and easy to adapt code...

If I just would be able to get more out of it... With the DeepStream SDK I was IMHO using a simplified four category FP16 resnet, which could make a pretty good bunch of the better performance. 

In my use case I just need to detect people and vehicles, but fps counts and it need to be three cams...

Would it be possible for you to give me a little wash sheet that would 

a) allow me to either run the minimized DeepStream model with your code or 
b) strip down one of the existing models?

TIA

dusty-nv, 2021-01-27:
Glad that you got it running fast - you could wrap up the deepstream stuff in Python modules to hide it.  I would stick with that if you have the performance you need.  The DeepStream models were trained with TLT (Transfer Learning Toolkit), and TLT does pruning, which is why the models are much faster (in addition to them having fewer classes and supporting batching).   Even if you were to re-train my SSD-Mobilenet on only cars+vehicles, the TLT model would likely still be faster (and more accurate because their models are production-quality).   However you could try re-training SSD-Mobilenet from [my tutorial](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md) by choosing those two classes from Google Open Images - beware it will be a lot of data unless you limit it (as described in the docs).   

I haven't yet added support for the TLT models in jetson-inference, so I would just stick with DeepStream for those.

neilyoung, 2021-01-28:
> you could wrap up the deepstream stuff in Python modules 

Even their Python sample code looks like as directly rewritten from C++. No structure, no beauty. Just crap...:(

closed 2023-04-27 by dusty-nv