dusty-nv/jetson-inference #903: training ssd-mobilenet from custom dataset (use CVAT and pascal voc)

2g1bz0, 2021-01-26:
i want to train my own dataset to detect emotions using ssd-mobilenet, iam using CVAT to label my images (pascal voc)
i put my dataset in ssd/data
command to run training: python3 train_ssd.py --dataset-type=voc --data=data/traffic_2 --model-dir=models/traffic_2 --epochs=100

output:
abc@abc-desktop:~/test/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=data/traffic_2 --model-dir=models/traffic_2 --epoch=100
2020-12-12 09:44:40 - Using CUDA...
2020-12-12 09:44:40 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/traffic_2', dataset_type='voc', datasets=['data/traffic_2'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=100, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-12-12 09:44:40 - Prepare training datasets.
warning - could not find image screenshot 2020-12-16 16.42.21 - ignoring from dataset
.....
warning - could not find image screenshot 2021-01-06 9.58.14 - ignoring from dataset
warning - could not find image screenshot 2021-01-06 9.58.51 - ignoring from dataset
2020-12-12 09:44:40 - No labels file, using default VOC classes.
2020-12-12 09:44:40 - Stored labels into file models/traffic_2/labels.txt.
2020-12-12 09:44:40 - Train dataset size: 0
Traceback (most recent call last):
  File "train_ssd.py", line 237, in <module>
    shuffle=True)
  File "/home/abc/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 224, in __init__
    sampler = RandomSampler(dataset, generator=generator)
  File "/home/abc/.local/lib/python3.6/site-packages/torch/utils/data/sampler.py", line 96, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integer value, but got num_samples=0

what should i do to solve it?
thank you.

dusty-nv, 2021-01-26:
Hi there- when you use CVAT, you need to make a labels.txt file, and after you extract your dataset put it inside your dataset's directory.

2g1bz0, 2021-01-26:
I created a labels.txt file and put it in the dataset's directory.
However, all screenshot data for training is ignored from the dataset.

abc@abc-desktop:~/test/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=data/traffic_2 --model-dir=models/traffic_2 --epoch=100
2020-12-12 08:03:14 - Using CUDA...
2020-12-12 08:03:14 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/traffic_2', dataset_type='voc', datasets=['data/traffic_2'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=100, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-12-12 08:03:14 - Prepare training datasets.
warning - could not find image screenshot 2020-12-16 16.42.21 - ignoring from dataset
warning - could not find image screenshot 2020-12-16 16.45.02 - ignoring from dataset
warning - could not find image screenshot 2020-12-16 16.45.20 - ignoring from dataset
...
2020-12-12 08:03:14 - VOC Labels read from file: ('BACKGROUND', 'speed_limit_50', 'speed_limit_40', 'speed_limit_30', 'crosswalk', 'stop', 'no_parking')
2020-12-12 08:03:14 - Stored labels into file models/traffic_2/labels.txt.
2020-12-12 08:03:14 - Train dataset size: 0
Traceback (most recent call last):
  File "train_ssd.py", line 237, in <module>
    shuffle=True)
  File "/home/abc/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 224, in __init__
    sampler = RandomSampler(dataset, generator=generator)
  File "/home/abc/.local/lib/python3.6/site-packages/torch/utils/data/sampler.py", line 96, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integer value, but got num_samples=0

dusty-nv, 2021-01-26:
Hmm.  What is the extension of your images?   Are the images stored under data/traffic_2/JPEGImages ?

2g1bz0, 2021-01-26:
sorry. The directory structure was incorrect.
The extension is png. Do I need to redo the annotation?
The images is stored under data/traffic_2/JPEGImages

```
- Annotations/
      - screenshot*.xml
- ImageSets/
      - Main
            - default.txt
- JPEGImages/
      - screenshot*.png
- labels.txt
- labelmap.txt
```

dusty-nv, 2021-01-27:
Hmm it should be able to load .png, here are the extensions supported:

https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/datasets/voc_dataset.py#L163

Is it able to load the images now?

2g1bz0, 2021-01-27:
The file name "Screenshot" was in my native language. Therefore, I changed the file name to English. Perhaps this was the cause.
Running the train is fine.
Thank you dusty

closed 2021-01-27 by 2g1bz0