dusty-nv/jetson-inference #905: Training Examples give different results than reported .

marksholcomb, 2021-01-27:
Wonderful examples and tutorials from Dusty! I have the jetson xavier nx and have followed the retraining Plant examples.  After 102 epoch I get  **Acc@1 46.608 Acc@5 79.383** after the "test" loop (example reported 55% TOP-1 and 85%-TOP5).  Curious if there is a 'tweak' I am missing or have scripts changed?  

I will download the PlantCLEF data set again and redo to insure I didn't follow the directions completely.

I have been training my own datasets and I find that my final recognition results seem poor, so, I am working through your larger data set example to isolate my errors.  I am new to this and hoping to learn.

dusty-nv, 2021-01-28:
Each training run may yield slightly different results from the psuedo-random initialization of weights and the random order that the training data is sampled by PyTorch - although I believe this difference is larger than expected.  

Are you using the jetson-inference container that has PyTorch 1.6 pre-installed?

Were you able to train the cat/dog model ok?    The PlantCLEF model is far more challenging, so I wouldn't sweat it too much.

marksholcomb, 2021-01-29:
Thanks Dusty.   It is working fine now.  I have been following Paul McWhorter videos and in a similar lesson we set the batch-size to 4, which I used again.  I upped it to the default batch-size of 8  , as is implied in your tutorial and repeat the accuracy you reported.  Is it be true that a lower Batch size requires a higher Epoch ? That was my experience anyhow.  I am new and will start digging deeper .. Thanks for your tutorials they are wonderful.

dusty-nv, 2021-01-29:
Hmm good to know - this blog seems to indicate that for much higher batch sizes, the training accuracy can actually decrease:

https://medium.com/mini-distill/effect-of-batch-size-on-training-dynamics-21c14f7a716e

Although those are for much higher batch sizes than we are dealing with.  Maybe with small batch sizes, it is the opposite phenomenon.

marksholcomb, 2021-01-29:
Fascinating stuff. Thanks for sharing that paper. I spent 40 years in aerospace structural optimization and it seems the same issues with gradient based algorithms, and finding the global optimum, apply to machine learning. Varying sample size and learning rate matter. Obvious to you machine learning guys, but , a eureka for me. I was going away from gradient based to genetic or doe based algorithms at the end of my working career. Faster computers made it practical, and you always find the  global optimum. Do design of experiments and genetic algorithms exist in machine learning space? Thanks for any clues !
Mark

dusty-nv, 2021-01-29:
Yea, there are genetic algorithms in machine learning for adapting hyperparameters and also evolution of the network architecture.  I haven't tried it myself and don't have specific papers to recommend, as it is a large field of research.

closed 2023-04-27 by dusty-nv