dusty-nv/jetson-inference #906: How to test the performance of a model using confusion matrix?

edwin9610, 2021-01-28:
Hello everyone,

I have trained a model using train_ssd.py script. I would like to test the performance of the model using confusion matrix. Since I am new to this terminology I am not sure how I can get the prediction and true dataset from the model. 

Can someone guide me through this or there is a better way to analyse the performance of the model?

Thank you.

dusty-nv, 2021-01-28:
Hi @edwin9610 , the [`eval_ssd.py`](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/eval_ssd.py) script could be used to evaluate the accuracy - although I have not used this script myself (it is from the upstream fork)

https://github.com/qfgaohao/pytorch-ssd#evaluation

edwin9610, 2021-01-31:
@dusty-nv I managed to evaluate the model accuracy by using your recommended script! Thank you!

closed 2021-01-31 by edwin9610

krishna-datascience, 2021-11-20:
@edwin9610 , you may please shared that script for evaluate the  model accuracy . Thanks!

Manoj-298, 2022-06-04:
@dusty-nv @edwin9610 @krishna-datascience  Hi sir, I try to evaluate the model using eval.py but I got an error please anyone tell me how I rectify this error

This is my error:
eval_1.py:69: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  all_gt_boxes[class_index][image_id] = torch.tensor(all_gt_boxes[class_index][image_id])
Traceback (most recent call last):
  File "eval_1.py", line 155, in <module>
    net.load(args.trained_model)
  File "/content/drive/MyDrive/Anamoly-project/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 135, in load
    self.load_state_dict(torch.load(model, map_location=lambda storage, loc: storage))
  File "/usr/local/lib/python3.7/dist-packages/torch/nn/modules/module.py", line 1498, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SSD:
	size mismatch for classification_headers.0.weight: copying a param with shape torch.Size([126, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 512, 3, 3]).
	size mismatch for classification_headers.0.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).
	size mismatch for classification_headers.1.weight: copying a param with shape torch.Size([126, 1024, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 1024, 3, 3]).
	size mismatch for classification_headers.1.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).
	size mismatch for classification_headers.2.weight: copying a param with shape torch.Size([126, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 512, 3, 3]).
	size mismatch for classification_headers.2.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).
	size mismatch for classification_headers.3.weight: copying a param with shape torch.Size([126, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 256, 3, 3]).
	size mismatch for classification_headers.3.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).
	size mismatch for classification_headers.4.weight: copying a param with shape torch.Size([126, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 256, 3, 3]).
	size mismatch for classification_headers.4.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).
	size mismatch for classification_headers.5.weight: copying a param with shape torch.Size([126, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([12, 256, 3, 3]).
	size mismatch for classification_headers.5.bias: copying a param with shape torch.Size([126]) from checkpoint, the shape in current model is torch.Size([12]).