dusty-nv/jetson-inference #909: how can I trans a uchar3* to three unit8_t* that split RGB to three unit8_t* by jetson-inference api ?

lzcchl, 2021-02-02:
hello, I want split a uchar3* (RGB) to three unit8_t* (RRRRR) (GGGGG) (BBBBB), is there any api in jetson-inference which use gpu not cpu?

lzcchl, 2021-02-03:
or Color conversion from (RGBRGBRGB......) TO (RRR...GGG...BBB...)

lzcchl, 2021-02-03:
or "uchar3*" Color conversion from (RGBRGBRGB......) TO (RRR...GGG...BBB...)

dusty-nv, 2021-02-03:
This conversion to NCHW planar format is done by the imageNet, detectNet, segNet classes during pre-processing, before the image is fed to the DNN.  Downsampling, mean-pixel subtraction and normalization are also applied.   These are done on the GPU and the functions are here:

https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorConvert.h

It's not really a public-facing API for doing it, although you could look at the CUDA code and make your own from it if all you want is the NCHW layout.

lzcchl, 2021-02-04:
thank you!!! I had not seen "tensorConvert.h" bofore because I think all API is in jetson-utils, "tensorConvert.h" works very well and speed my project.

closed 2021-02-07 by lzcchl