dusty-nv/jetson-inference #910: Stream cudaimage over RTP

apoorvmishra28, 2021-02-04:
Hi @dusty-nv 

I'm trying to stream the processed frames from detectnet.py over RTP using GStreamer. Can you suggest me a way to do that without using OpenCV?

dusty-nv, 2021-02-04:
Hi @apoorvmishra28 , you can output RTP using command-line to detectnet like this:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#transmitting-rtp

apoorvmishra28, 2021-02-05:
Thank you for the quick response @dusty-nv 
yes, I can transmit the RTP using GStreamer but can you please suggest me a way for receiving that RTP stream with a single link after installing the GStreamer.
like, suppose I want to stream directly on the front end application ex: reactJS.

dusty-nv, 2021-02-05:
I haven't integrated the RTP stream into a web page or web app before, but I just found this project that integrates GStreamer with reactJS:

https://github.com/Kalyzee/react-native-gstreamer

So perhaps you could use that project and a GStreamer pipeline like this:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#viewing-rtp-remotely

apoorvmishra28, 2021-02-17:
Hey @dusty-nv ,
I tried streaming video frames via RTP and it is working fine with video-viewer. I'm able to get the streams via video-viewer very easily but it is not working with SDP file. Below is the SDP file content I've used.

  ```
c=IN IP4 MY_IP
  m=video 5000 RTP/AVP 96
  a=rtpmap:96 H264/90000
```
Am I missing out on anything? Please let me know.

dusty-nv, 2021-02-17:
Hi @apoorvmishra28, hmm it appears to be fine, assuming you want it on port 5000 and your encoding codec is still H.264.

If you monitor the IP packets being received on the destination computer (i.e. with ifconfig or ipconfig), do the packets go up when the stream is being sent?   Otherwise there may be some firewall issue.

You can also test if GStreamer on the destination computer is able to receive it.  I have found sometimes VLC can be finicky with RTP, whereas using GStreamer to receive the RTP feed is much more reliable.  You should be able to install GStreamer on Windows/Mac/Linux if that's an option for you.  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#rtp

apoorvmishra28, 2021-02-19:
Hi @dusty-nv , Thanks a lot for your quick response. I'm able to read the RTP stream via GStreamer.

I'd also tried streaming the raw frames via openCV + GStreamer over to the **FFMPEG Server**, provided below is the command that I've used to generate  RTSP stream:
```
std::string pipe = "appsrc ! video/x-raw, format=BGR ! queue ! videoconvert \
						! video/x-raw, format=BGRx ! nvvidconv ! omxh264enc \
						! video/x-h264, stream-format=byte-stream ! h264parse \
						! rtph264pay pt=96 config-interval=1 ! \
						udpsink host=MY_IP port=5000  sync=false async=false";
```
So, can you please suggest me a way to send stream over to the **FFMPEG server**.

apoorvmishra28, 2021-02-27:
Hi @dusty-nv , Thanks for your support. I'd manage to stream the video to RTMP server using opencv FFMPEG and subprocess. I'd used:

```
ffmpeg -y -f rawvideo -vcodec rawvideo -pix_fmt bgr24 -s 640x480 -i - -profile baseline -b 100k -c:v libx264 \
                           -pix_fmt yuv420p  \
                           -preset slow -r 40 -f flv rtmp_url
```
I hope someone would find this helpful.

closed 2021-02-27 by apoorvmishra28

neilyoung, 2021-02-27:
Yes, I find it helpful :) What is the latency you observe at the RTMP input level?

apoorvmishra28, 2021-02-28:
Hi @neilyoung , I'd observed a latency of around 200.

neilyoung, 2021-02-28:
200 ms with RTMP?