dusty-nv/jetson-inference #911: Poor performance of pre-trained segnet model

sukhrajklair, 2021-02-05:
[Semantic segmentation gif](https://giphy.com/gifs/hbrwVAWj0GVBJOtkCZ/html5)

I appreciate for sharing your work. 

I am new to this and trying to use the pre-trained model **fcn-resnet18-cityscapes-2048x1024** to do real-time semantic segmentation for a robotics application. I am also using your [ros repo](https://github.com/dusty-nv/ros_deep_learning) to implement the pipeline. 

The input is a simulated 800x800 camera located inside an urban environment simulated in gazebo. The frame rate is less than 20fps. As you can see in the above liked gif, a lot of pixels are not labelled properly. The sidewalk is not recognized at all. 

Is this an expected behavior or is there something I missed? How do I improve the labeling?

dusty-nv, 2021-02-05:
You might want to try the smaller `fcn-resnet18-cityscapes-512x256` model, because your 800x800 image is being upsampled to 2048x1024 for the 2048x1024 model.  Also, you may want to change/crop the aspect ratio to be the same as the model.  I have also not tried this on simulated data, only real images.

sukhrajklair, 2021-02-07:
I made those changes and it works significantly better. I’ll try with real images to see if it’s any better with them. 

Thank you so much.

closed 2021-02-07 by sukhrajklair