dusty-nv/jetson-inference #912: High Resolution Object Detection Model

rsamvelyan, 2021-02-07:
Hello
I'm looking for an SSD object detection model that has a higher resolution than MobileNet. Ideally something like 1200x1200.
Is there a way to increase the MobileNet's resolution or is there another model which I could run inference on Jetson Xavier and which would be supported by DetectNet? 
Thanks!

dusty-nv, 2021-03-01:
Hi @rsamvelyan, you might want to see this forum thread where we got SSD-512 working:

https://forums.developer.nvidia.com/t/how-train-jetson-inference-ssd512-model/168510/6?u=dusty_nv

rsamvelyan, 2021-03-02:
Thanks a lot Dusty! I will try to do it as well. 
I wonder if SSD-512 could work with MobileNet V1?  Or is it going to have an incompatible size? 
I just need it to run real-time.

closed 2023-04-27 by dusty-nv