dusty-nv/jetson-inference #913: python api not able to load custom ssd v2

rlrahulkanojia, 2021-02-09:
I have a custom model that i am able to run using command line like 
`detectnet --model=<path_to_my_model.onnx>, `

but not able to load model using python api 

`jetson.inference.detectNet(<path_to_my_model.onnx>, DETECTION_THRESHOLD)' 

error -  "jetson.inference -- detectNet invalid built-in network was requested ('/home/abs/Documents/ONNX/mb2-ssd-lite.onnx')"

rlrahulkanojia, 2021-02-10:
Got it working, passed the aruguments as a list in sys.argv

jetson.inference.detectNet(<path_to_my_model.onnx>, ['models' <path to custom models] ,DETECTION_THRESHOLD)'

closed 2021-02-10 by rlrahulkanojia

surefyyq, 2021-04-15:
Thanks for your share!
But what's the meaning of  `['models' <path to custom models]`, Is it the same as the path to .onnx file?