dusty-nv/jetson-inference #914: Training SSD-Inception-v2 model

philnguyenresson, 2021-02-09:
Hi, is there a way to train a SSD-Inception-v2 model which would work with jetson-inference?
It seems that the training code in the docs handles all the other models except SSD-Inception-v2.

Similarly, reading some of the other issue pages, it seems like people have had success getting models not mentioned in the docs to work. Is there a an external source for potential models?

dusty-nv, 2021-02-09:
Hi @philnguyenresson , you would need to modify the pytorch-ssd code to support Inception-v2 backbone with SSD.   That SSD-Inception model that comes with the repo came from TensorFlow and converted to TensorRT UFF using [this tool](https://github.com/AastaNV/TRT_object_detection).  The PyTorch/ONNX-based workflow is much easier to use.

Getting external models to work typically requires getting the pre/post-processing to match what is done in PyTorch/TensorFlow.  And if the model is exported to ONNX for example, TensorRT needs to be able to load the ONNX.

philnguyenresson, 2021-02-09:
Ok, that makes sense thanks

closed 2021-02-09 by philnguyenresson