dusty-nv/jetson-inference #915: docker/run.sh driver error

emrectn, 2021-02-09:
Hi, I follow the [installation](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) guide. When I run the docker/run.sh command, there is an error that it following below. Could you help me please ?

```bash
nvidia@Jetson:~/Documents/jetson-inference$ docker/run.sh
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.5.0
size of data/networks:  535209937 bytes
CONTAINER:     dustynv/jetson-inference:r32.5.0
DATA_VOLUME:   --volume /home/nvidia/Documents/jetson-inference/data:/jetson-inference/data --volume /home/nvidia/Documents/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/nvidia/Documents/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/nvidia/Documents/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/nvidia/Documents/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
docker: Error response from daemon: OCI runtime create failed: container_linux.go:370: starting container process caused: process_linux.go:459: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver error: failed to process request: unknown.
```

emrectn, 2021-02-09:
I share some outputs for information purposes. 

```bash
nvidia@Jetson:~/Documents/jetson-inference$ dpkg -l | grep jetson
ii  jetson-gpio-common                         2.0.11                                           arm64        Jetson GPIO library package (common files)
ii  nvidia-l4t-jetson-io                       32.5.0-20210115151051                            arm64        NVIDIA Jetson.IO debian package
ii  nvidia-l4t-jetson-multimedia-api           32.5.0-20210115151051                            arm64        NVIDIA Jetson Multimedia API is a collection of lower-level APIs that support flexible application development.
ii  python-jetson-gpio                         2.0.11                                           arm64        Jetson GPIO library package (Python 2)
ii  python3-jetson-gpio                        2.0.11                                           arm64        Jetson GPIO library package (Python 3)
```

```bash
nvidia@Jetson:~/Documents/jetson-inference$ dpkg -l | grep cudnn
ii  libcudnn8                                  8.0.0.180-1+cuda10.2                             arm64        cuDNN runtime libraries
ii  libcudnn8-dev                              8.0.0.180-1+cuda10.2                             arm64        cuDNN development libraries and headers
ii  libcudnn8-doc                              8.0.0.180-1+cuda10.2                             arm64        cuDNN documents and samples
ii  nvidia-container-csv-cudnn                 8.0.0.180-1+cuda10.2                             arm64        Jetpack CUDNN CSV file
```

dusty-nv, 2021-02-09:
Hmm... can you output `sudo docker info` ?

Are you able to start l4t-base?

``` bash
sudo docker run --runtime nvidia -it --rm --network host nvcr.io/nvidia/l4t-base:r32.5.0
```

emrectn, 2021-02-10:
Hi dusty, you can look output.

```bash
nvidia@Jetson:~$ sudo docker info
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Build with BuildKit (Docker Inc., v0.5.1-docker)

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 2
 Server Version: 20.10.3
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runtime.v1.linux nvidia runc io.containerd.runc.v2
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc version: ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 init version: de40ad0
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.9.201-tegra
 Operating System: Ubuntu 18.04.5 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 7.666GiB
 Name: Jetson
 ID: KHTT:VUZR:CZPP:WEWO:NENZ:2LRA:Z3K2:67XZ:A7UV:S7TA:DHH4:T7C6
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No blkio weight support
WARNING: No blkio weight_device support
```

```bash
nvidia@Jetson:~$ sudo docker run --runtime nvidia -it --rm --network host nvcr.io/nvidia/l4t-base:r32.5.0
Unable to find image 'nvcr.io/nvidia/l4t-base:r32.5.0' locally
r32.5.0: Pulling from nvidia/l4t-base
5000a6c32c5a: Pull complete 
8e855b69096a: Pull complete 
8db8dbbd4bb9: Pull complete 
833dc3235950: Pull complete 
f79d264135a3: Pull complete 
1c40f77bb35b: Pull complete 
1990ecf0bfb7: Pull complete 
c8ffbfd7f0aa: Pull complete 
ba785779122a: Pull complete 
024ce79b6790: Pull complete 
9b09da3b5483: Pull complete 
17f974a43cf9: Pull complete 
211b56b73ff1: Pull complete 
78aca4be1f3b: Pull complete 
95f34310bbda: Pull complete 
678c9d1557e9: Pull complete 
ec17ad7cab01: Pull complete 
08fb1eee5328: Pull complete 
Digest: sha256:c7207a13da6054c738b530aad93e8c620182db85eb7fbb40b103757145e8b5f4
Status: Downloaded newer image for nvcr.io/nvidia/l4t-base:r32.5.0
docker: Error response from daemon: OCI runtime create failed: container_linux.go:370: starting container process caused: process_linux.go:459: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver error: failed to process request: unknown.

```

dusty-nv, 2021-02-10:
Hmm you are unable to start the `l4t-base` container too, so the issue isn't specific to the `jetson-inference` container.   I think something got messed up with your nvidia-docker setup but I'm not sure what.  You may want to re-install nvidia-docker or reflash JetPack.

However, if you just want to run jetson-inference project and don't care about Docker, you can just [`Build from Source`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) outside of Docker.

emrectn, 2021-02-12:
Hello, I had already Jetson 4.4 and used l4t-base:32.4.0 so I thought something got messed up with my nvidia-docker setup. And I reinstall jetson and nvidia docker. I Upgraded jetson 4.4 to 4.5 but I have same error. Could you give any idea about this ?  Anyone running nvidia docker on a specific Jetson version?

dusty-nv, 2021-02-12:
Something must have happened during the upgrade - I think you might just want to re-flash your SD card with JetPack 4.5 image (or flash a different SD card with JetPack 4.5).

emrectn, 2021-02-15:
Hi dusty, I downgrade jetpack 4.4. So I can run this command docker/run.sh. Thank you so much. But there is a problem when run followign command.

```bash
nvidia@Nvidia:~/Desktop/jetson-inference$ sudo docker run --rm --gpus all nvidia/cuda:10.2-base nvidia-smi
docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "process_linux.go:449: container init caused \"process_linux.go:432: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: src: /usr/lib/aarch64-linux-gnu/libnvinfer.so.7, src_lnk: libnvinfer.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvinfer.so.7, dst_lnk: libnvinfer.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.7, src_lnk: libnvinfer_plugin.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.7, dst_lnk: libnvinfer_plugin.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvparsers.so.7, src_lnk: libnvparsers.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvparsers.so.7, dst_lnk: libnvparsers.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvonnxparser.so.7, src_lnk: libnvonnxparser.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvonnxparser.so.7, dst_lnk: libnvonnxparser.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvinfer.so, src_lnk: libnvinfer.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvinfer.so, dst_lnk: libnvinfer.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so, src_lnk: libnvinfer_plugin.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so, dst_lnk: libnvinfer_plugin.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvparsers.so, src_lnk: libnvparsers.so.7.1.3, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvparsers.so, dst_lnk: libnvparsers.so.7.1.3\\\\nsrc: /usr/lib/aarch64-linux-gnu/libnvonnxparser.so, src_lnk: libnvonnxparser.so.7, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libnvonnxparser.so, dst_lnk: libnvonnxparser.so.7\\\\nsrc: /usr/lib/aarch64-linux-gnu/libmyelin.so.1, src_lnk: libmyelin.so.1.1.0, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libmyelin.so.1, dst_lnk: libmyelin.so.1.1.0\\\\nsrc: /usr/lib/aarch64-linux-gnu/libmyelin.so, src_lnk: libmyelin.so.1, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libmyelin.so, dst_lnk: libmyelin.so.1\\\\nsrc: /usr/lib/libvisionworks_sfm.so, src_lnk: libvisionworks_sfm.so.0.90, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/libvisionworks_sfm.so, dst_lnk: libvisionworks_sfm.so.0.90\\\\nsrc: /usr/lib/libvisionworks_sfm.so.0.90, src_lnk: libvisionworks_sfm.so.0.90.4, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/libvisionworks_sfm.so.0.90, dst_lnk: libvisionworks_sfm.so.0.90.4\\\\nsrc: /usr/lib/libvisionworks.so, src_lnk: libvisionworks.so.1.6, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/libvisionworks.so, dst_lnk: libvisionworks.so.1.6\\\\nsrc: /usr/lib/libvisionworks_tracking.so, src_lnk: libvisionworks_tracking.so.0.88, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/libvisionworks_tracking.so, dst_lnk: libvisionworks_tracking.so.0.88\\\\nsrc: /usr/lib/libvisionworks_tracking.so.0.88, src_lnk: libvisionworks_tracking.so.0.88.2, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/libvisionworks_tracking.so.0.88, dst_lnk: libvisionworks_tracking.so.0.88.2\\\\nsrc: /usr/lib/aarch64-linux-gnu/libcudnn.so.8, src_lnk: libcudnn.so.8.0.0, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libcudnn.so.8, dst_lnk: libcudnn.so.8.0.0\\\\nsrc: /usr/lib/aarch64-linux-gnu/libcudnn.so, src_lnk: /etc/alternatives/libcudnn_so, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libcudnn.so, dst_lnk: /etc/alternatives/libcudnn_so\\\\nsrc: /usr/lib/aarch64-linux-gnu/libcudnn_ops_infer.so.8, src_lnk: libcudnn_ops_infer.so.8.0.0, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libcudnn_ops_infer.so.8, dst_lnk: libcudnn_ops_infer.so.8.0.0\\\\nsrc: /usr/lib/aarch64-linux-gnu/libcudnn_ops_infer.so, src_lnk: /etc/alternatives/libcudnn_ops_infer_so, dst: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/lib/aarch64-linux-gnu/libcudn [... cut ...] rand=tesla,driver>=396,driver<397 brand=tesla,driver>=410,driver<411 brand=tesla,driver>=418,driver<419 brand=tesla,driver>=440,driver<441 --pid=12738 /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged]\\\\nnvidia-container-cli: mount error: stat failed: /var/lib/docker/overlay2/1251d7a31872e130ae470fe6ac106f62d005d27d0e201292069f354c3e3dec94/merged/usr/local/cuda-10.2/compat/libcuda.so.440.118.02: no such file or directory\\\\n\\\"\"": unknown.

```

dusty-nv, 2021-02-15:
That nvidia/cuda:10.2-base container is for x86, not Jetson/aarch64.   The equivalent on Jetson is l4t-base

emrectn, 2021-02-16:
Okay, l4t-base:32.4 have cuda 10.2 version. I have to use tensorflow==2.2 to my application. Bu the version of tensorflow support CUDA 10.0. How can I solve this problem ? Can I downgrade Cuda version 10.2 to 10.0 ?

dusty-nv, 2021-02-16:
You can use TensorFlow 2.3?   You could use the l4t-tensorflow container:   https://ngc.nvidia.com/catalog/containers/nvidia:l4t-tensorflow

The only way to downgrade CUDA is to go back to a previous JetPack version, which would be JetPack 4.4 (L4T R32.4.3) for TensorFlow 2.2.

emrectn, 2021-02-17:
Thank you dusty

closed 2021-02-17 by emrectn