dusty-nv/jetson-inference #917: HELP: Editing/running python code in docker jetson-inference contanier with VS code

mmatesic01, 2021-02-10:
Hi, 
I am trying to open/create code in docker jetson-inference in VC code, but I am not ther taht I coudi edit/create and run python code contained in Docker.
I have installed Docker and Remote contanier extension VS Code, but I am really serching the propper way to do this.
Any advice?
Thank you.

dusty-nv, 2021-02-10:
Hi @mmatesic01 , it's recommended to put your Python code in a docker mounted directory, so your code persists outside of the container.  The [`docker/run.sh`](https://github.com/dusty-nv/jetson-inference/blob/master/docker/run.sh) script takes a `--volume` argument that allows you to mount a directory from the host into the container.   For example:

``` bash
$ mkdir ~/my_code
$ touch ~/my_code/my_script.py
$ docker/run.sh --volume ~/my_code:/my_code
```

Your code will then show up at `/my_code` inside the container, and you can edit it from outside the container.

mmatesic01, 2021-02-13:
Hi dusty,
I am getting closer to what I want. I corected the commands above to my environment.
And I can see the listed file in my container in my VS code. But when I open it it is shown as empty and read-only, altough file outside container has contents.
[image]

closed 2023-04-27 by dusty-nv