dusty-nv/jetson-inference #919: How to mark my own images without using camera-capture?

jugfk, 2021-02-12:
Dear Dusty

This is Stephen Lee.
**I want to mark my own images without using camera-capture.**
So, the information would be XML data with bound box information
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md

dusty-nv, 2021-02-12:
Hi @jugfk, please see my reply to similar question here:

https://forums.developer.nvidia.com/t/camera-capture-for-single-images-inside-a-local-folder/168317/2?u=dusty_nv

jugfk, 2021-02-12:
That is not the question.

dusty-nv, 2021-02-12:
The question was 'how can I annotate images that already exist', which is the same as yours.  You can use the CVAT tool to do that.

jugfk, 2021-02-12:
@dusty-nv I have 100 images without using the camera-capture. I have succeeded in the experiment of Training Image Classification Models. but I'm struggling with Training Object Detection Models.

dusty-nv, 2021-02-12:
Import your images into CVAT, annotate them with bounding boxes, and export from CVAT in Pascal VOC format.  Then add a labels.txt to your extracted dataset, where there is one classname per line of your labels.txt.  Then train it with train_ssd.py like you would have. 

You may need more than 100 images to get a good result, it depends on the subject matter / environment conditions and the accuracy you desire.

jugfk, 2021-02-12:
> The question was 'how can I annotate images that already exist', which is the same as yours. You can use the CVAT tool to do that.

@dusty-nv  Thank you for precious time. I'm also looking for offline software. Could you recommend it?

dusty-nv, 2021-02-12:
I believe you can install CVAT locally, although I haven't done that.  I had tried a couple other tools but CVAT exported the data in the closest way.

jugfk, 2021-02-13:
> I believe you can install CVAT locally, although I haven't done that. I had tried a couple other tools but CVAT exported the data in the closest way.

Thank you for helping. I have done to locally install it.

closed 2021-02-13 by jugfk