dusty-nv/jetson-inference #922: Segmentation mask output in segnet, how to convert to cv::Mat (in C++)

wmuron, 2021-02-15:
Hi, 

I would like to apply some custom post-processing using OpenCV (C++) methods on the output mask from segNet. Is there any documentation or examples on how to convert/reinterpret `uchar3 imgMask` to a `cv::Mat` object? To clarify, I only need to postprocess the mask inside the loop (e.g. here: https://github.com/dusty-nv/jetson-inference/blob/master/examples/segnet/segnet.cpp#L289).

I can see that there are python-cv2 to/from numpy utilities, but unfortunately I need to keep the implementation in C++. I would really appreciate if you could share any hints/code examples on how to approach this. Thanks!

dusty-nv, 2021-02-15:
Hi @wmuron , does something like this work?

``` cpp
CUDA(cudaDeviceSynchronize());
cv::Mat cv_image(cv::Size(imgWidth, imgHeight), CV_8UC3, imgMask);
```

I think you may need to convert from RGB->BGR with something like `cv::cvtColor(cv_image, cv_image2, cv::COLOR_RGB2BGR);`.  Or cudaConvertColor() from jetson-utils can do it beforehand on the GPU.

Also if you want the class ID's instead of colors, you may want to look at the uint8 version of the segNet::Mask() function, which outputs class ID's:   https://github.com/dusty-nv/jetson-inference/blob/f7cdfad059acd3d300acaec1f312ede27c049689/c/segNet.h#L256

wmuron, 2021-02-16:
Thank you for your response, the line works perfectly. Also, I've switched to uint8 version of segNet::Mask(), definitely makes more sense in my case.

closed 2021-02-24 by wmuron

mprabhakarece, 2021-06-03:
@wmuron Please tell how to enable opencv libraries for above mentioned problem.