dusty-nv/jetson-inference #923: Custom Rendering

ahndroo, 2021-02-16:
I'm working with the detectNet model in one of my applications.  After performing using the .Detect method, is there a way to clear the detection attribute?  

I'm trying to do my own custom rendering and when I plot using the display.Render(image) method it displays my changes along with the jetson-inference overlays.  When I initialized the videoOutput, I sent it the --overlay none flag and it helped but now every once in a while I'll get a small bounding box on the screen where it's trying to use the jetson-inference rendering tool.  

Is there a way to delete the detection history so the rendering stops trying to grab that information in the display?

dusty-nv, 2021-02-17:
Hi @ahndroo, the detectNet.Detect() function takes an optional overlay argument, that you can set to `none` and it won't overlay the bounding boxes/labels:

``` python
net.Detect(img, overlay='none')
```

Does that work for you?

JNaranjo-Alcazar, 2022-12-12:
Oh great! 

thank you very much for your quick response. I haven't been able to test the project yet because I'm waiting for the NVIDIA device and I was having some doubts before getting down to work.

Once again, thank you very much.

dusty-nv, 2022-12-12:
No problem @JNaranjo-Alcazar!  Good luck with your project when you receive your device!

BTW there are some examples of working directly with `jetson.utils.cudaImage` objects here:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md
* https://github.com/dusty-nv/jetson-utils/tree/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/python/examples

closed 2023-04-27 by dusty-nv