dusty-nv/jetson-inference #924: Can I delete the training pictures after I exported the ONNX?

shanhwa, 2021-02-17:
Mr.Dusty,

After I export the ONNX file, does it mean that I can delete the training pictures, such as the picture files in cat_dog.

dusty-nv, 2021-02-17:
Hi @shanhwa, yes if you are done training, you can delete the training pictures.

shanhwa, 2021-02-18:
Do I just need to keep the ONNX file? Can all other jpg files be deleted?

BTW, OMG! I'm your big fan, Mr.Dusty!

dusty-nv, 2021-02-18:
Haha, thanks!   Keep the ONNX file, the labels.txt, and it is good to keep the PyTorch checkpoints (or at least the PyTorch checkpoint that was the best).   The training dataset you can delete if you are done with it.

shanhwa, 2021-02-19:
Thank you Mr.Dusty,

The following are the directories I will keep and the directories I will delete. Can you see if they are correct?

classification/
data/
  |-cat_dog --delete
    |-test --delete
    |-test_cat_output --delete
    |-test_dog_output --delete
    |-train --delete
    |-val --delete
    |-labels.txt --keep
models/cat_dog/
  |-checkpoint.pth.tar --delete
  |-model_best.pth.tar --keep
  |-resnet18.onnx --keep
  |-resnet18.onnx.1.1.7103.GPU.FP16.engine --keep

Look forward to your reply, Mr.Dusty.

neilyoung, 2021-02-19:
@dusty-nv OT OT OT: I'm following now your repo since a month and I am totally impressed with how you are managing this. The dedication and quality that users of your software can expect sets the bar. Always ready with the answer and rarely anything left unanswered. I would like to see something similar at DeepStream, where I migrated on your advice. Technically quite reasonable I think, but from the support underground bad. Not even remotely comparable to what is offered here and actually not even in the slightest the good Nvidia standard hitting. Too bad, the right people always work in the wrong places :)

Keep up the good work Dusty. You are really a nugget on the Internet

dusty-nv, 2021-02-19:
> classification/
> data/
> |-cat_dog --delete
> |-test --delete
> |-test_cat_output --delete
> |-test_dog_output --delete
> |-train --delete
> |-val --delete
> |-labels.txt --keep
> models/cat_dog/ [...]

Hi @shanhwa, that is correct - note that the tutorial uses the dataset for testing the model (i.e. running the images in `cat_dog/test`), but it appears you have already done that testing.  So you can go ahead and delete above, and you will still be able to run the model (i.e. on other images you may have supplied, or a camera stream).   And worst case, you can just download/extract the cat_dog dataset again.

> I'm following now your repo since a month and I am totally impressed with how you are managing this. The dedication and quality that users of your software can expect sets the bar. Always ready with the answer and rarely anything left unanswered.

Thanks, always happy to help!

shanhwa, 2021-02-20:
Thank you very much Mr.Dusty.

closed 2021-02-20 by shanhwa