dusty-nv/jetson-inference #927: Retrain SSD-Mobilenet non-labeled objects

stekla79, 2021-02-18:
I have retrained a SSD-Mobilenet model with good results. However in certain situations I get false positives with objects kind of resembling the objects trained but there is a distinct difference. 

Is there a way to include non-labeled objects like negative samples without bounding boxes to approve accuracy? 

I can do this with darknet and get really good results but havn't found a way to do this with mobilenet. Is it possible and if so, how should I structure the data?

dusty-nv, 2021-02-18:
Hi @stekla79, pytorch-ssd will sample from negative areas to build up the negative examples, but I think each image requires at least one positive bounding box.

stekla79, 2021-02-19:
Thanks for the reply. It's hard to collect samples with the false positives alongside the true positives. 

Maybe some photoshop and insertion of false positive in an image with a true positive could work. This would be a lot of manual work taking some time though.

dusty-nv, 2021-02-19:
Perhaps the code could also be modified to work with images that had 0 annotations.  I had to manually disable images with 0 annotations for VOC-type datasets, because it would throw an exception later in the code:

https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/datasets/voc_dataset.py#L117

But perhaps a check later in the code could use 0-annotation images only for negative examples.  This pytorch-ssd code was forked from an upstream source.

alvaro-cmd, 2022-07-08:
That would be appreciated, I'm having the same problem.

alvaro-cmd, 2022-07-08:
If I create a label called "BACKGROUND" and simply annotate the whole image as a BACKGROUND detection. Would it be interpreted as a negative example?

closed 2023-04-27 by dusty-nv