dusty-nv/jetson-inference #928: Can someone tell me which Jeetson Device is being referred to in this repo when they say "Xavier"

sisaha9, 2021-02-18:
https://github.com/dusty-nv/jetson-inference#semantic-segmentation

I know Jetson Nano but which Xavier is being referred to here? Xavier NX or AGX Xavier?

dusty-nv, 2021-02-18:
That was with Jetson AGX Xavier

https://github.com/dusty-nv/jetson-inference#semantic-segmentation

sisaha9, 2021-02-18:
Which semantic segmentation model would you recommend for real time inference in Xavier NX?

sisaha9, 2021-02-18:
I have 720x1280 images but I will scale down if need be since fast real time inference is important for my application (self driving car)

dusty-nv, 2021-02-18:
They should all still be realtime on NX.  If your input is 1280x720, try `fcn-resnet18-cityscapes-1024x512` or `fcn-resnet18-cityscapes-512x256`

sisaha9, 2021-02-18:
Got it. Thanks!

closed 2021-02-18 by sisaha9