dusty-nv/jetson-inference #929: Detection training

Sobryx, 2021-02-18:
Hi, I have a question that might seem stupid but there is something I didn't quite get. I had no knowledge or experience of machine learning beforehand.

So I followed the tutorial and created my own model with the person class.

**When I train this model everything goes well but something is bothering me. What exactly is the argument "--resume=..." doing ?**

What I understood is that it is using the the save of the model I chose so using the save weights to start over from a much better start and then improving faster.  Is this assumption correct ? If not may I have an explanation ?

I've been training and observing the trend of validation loss at each epoch and when it was going up I was aborting the training and starting over. However when it was going down I was letting it continue and then using the new best model to retrain using --resume. Problem is that most of the time the loss is going up and the amelioration is negligeable when it happens.

For exemple I started off with 3.93 validation loss and after something like 150 epochs overall, it went down to 3.55 for my best save. Does that seem any good ? I mean I noticed differences when I runned it on a test video but it's still far from perfect.

**My other question is about the argument "--pretrained-ssd". Is this the same as "--resume" ? If not what is the difference ?**

Thanks for reading and I look forward to any answer I can get. 

Have a good day!

dusty-nv, 2021-02-18:
Your concept is correct, since training can take awhile you can use `--resume` or `--pretrained-ssd` to restart the training.  If `--resume` is giving you problems, then try `--pretrained-ssd` instead.  I haven't quite figured out why sometimes loss goes up on `--resume`, I think it may be that the annealed learning-rate schedule gets reset back to the base learning rate (annealed meaning that the learning rate typically decays over time).

The level of loss seems to depend on the dataset, for example if it's a challenging dataset the loss can be higher while still producing a good result.

Sobryx, 2021-02-18:
Thanks for that fast answer!

So i'm better off using the `--pretrained-ssd` argument ?

[image]

About that lr_scheduler warning I get when I start training. Should I be worried about that ?

dusty-nv, 2021-02-18:
If you are having problems with `--resume`, then try `--pretrained-ssd`.  You can ignore that error (although I should look into fixing that)

Sobryx, 2021-02-18:
Allright, thanks !

closed 2021-02-18 by Sobryx