dusty-nv/jetson-inference #931: low accuracy for detect far objects

shaojun, 2021-02-19:
Hi dusty,
thanks for the great repo.
I'm using the `ssd-mobilenet-v2` as in the repo's example for a detection demo for the team here, noticed if i put the object a bit far, the detection accuracy will drop significantly.
also, I've tried change the network to `SSD-Inception-v2`, even the different camera hardware(logictech c270 and c920), simliar behavior.
do you have any suggestion?

dusty-nv, 2021-03-01:
Hi @shaojun, you might want to see this forum thread where we got SSD-512 working:

https://forums.developer.nvidia.com/t/how-train-jetson-inference-ssd512-model/168510/6?u=dusty_nv

It can help find smaller objects (i.e. farther away)

shaojun, 2021-03-06:
thanks dusty.
Will follow that to train a 512 ssd on mobile net.
if increase the training resolution for higher, like 720p even 1080p, what's the pros and cons? and why the default sample training script using such a low resolution of 300x300 .

dusty-nv, 2021-03-06:
Resolution of 300x300 (or more commonly 224x224) is frequent the default for DNNs using backbones like ResNet/Alexnet/GoogleNet/Mobilenet/ect.  They don't typically need HD resolution like other algorithms.  If you are trying to detect very small objects, then 512x512 could help.

I haven't tried greater than 512 with pytorch-ssd code, it would require adjusting the anchors and I'm not sure how those ratios are arrived at.  We got those for 512 from another thread on the upstream github repo.

xmba15, 2021-05-12:
@dusty-nv @shaojun 
after looking for methods to get default boxes' spec for larger resolution, I came up with my own solution in this branch, reading the paper and the author's code.
https://github.com/xmba15/pytorch-ssd/commit/bc0d3cccf49a7342010b3e42831bb97ac8ad6ff1
The model gives me much better accuracy with higher resolution.

xmba15, 2021-05-12:
@dusty-nv I think the following specs in 
https://github.com/dusty-nv/pytorch-ssd/blob/res-512/vision/ssd/config/mobilenetv1_ssd_config.py
are not correct. at least for the shrinkage column (second column) (16,32,64,100,150,300)

as 
```
shrinkage (or step) = image size / feature map size
```
the second column should be 512/32,512/16,512/8,512/4,512/2,512/1 --> 16,32,64,128,256,512

alessandrolombardini, 2021-05-19:
> @dusty-nv @shaojun
> after looking for methods to get default boxes' spec for larger resolution, I came up with my own solution in this branch, reading the paper and the author's code.
> [xmba15/pytorch-ssd@bc0d3cc](https://github.com/xmba15/pytorch-ssd/commit/bc0d3cccf49a7342010b3e42831bb97ac8ad6ff1)
> The model gives me much better accuracy with higher resolution.

How did you do?

xmba15, 2021-05-24:
To use input of larger resolution (for better accuracy), you need to modify the specs of anchor boxes of SSD (in case you are not familiar with the algorithm, you can understand the specs as the algorithm's intrinsic parameters). What I did is to write a script to generate those specs automatically for any feature-extractor backbones.

xmba15, 2021-05-24:
@alessandrolombardini @dusty-nv 
for example, here is the specs for input size of 512 x 512
```python
image_size = 512

SSDSpec(32, 16, SSDBoxSizes(51.2, 102.4), [2, 3]),
SSDSpec(16, 32, SSDBoxSizes(102.4, 189.44), [2, 3]),
SSDSpec(8, 64, SSDBoxSizes(189.44, 276.48), [2, 3]),
SSDSpec(4, 128, SSDBoxSizes(276.48, 363.52), [2, 3]),
SSDSpec(2, 256, SSDBoxSizes(363.52, 450.56), [2, 3]),
SSDSpec(1, 512, SSDBoxSizes(450.56, 537.6), [2, 3]),
```

and specs for image_size of 608x608

```python
image_size = 608

specs = [
    SSDSpec(38, 16, SSDBoxSizes(45.6, 91.2), [2, 3]),
    SSDSpec(19, 32, SSDBoxSizes(91.2, 200.64), [2, 3]),
    SSDSpec(10, 61, SSDBoxSizes(200.64, 310.08), [2, 3]),
    SSDSpec(5, 122, SSDBoxSizes(310.08, 419.52), [2, 3]),
    SSDSpec(3, 203, SSDBoxSizes(419.52, 528.96), [2, 3]),
    SSDSpec(2, 304, SSDBoxSizes(528.96, 638.4), [2, 3]),
]
```
change those specs params in https://github.com/dusty-nv/pytorch-ssd/blob/res-512/vision/ssd/config/mobilenetv1_ssd_config.py

alessandrolombardini, 2021-05-30:
Very very nice, thank you very much.

closed 2023-04-27 by dusty-nv