dusty-nv/jetson-inference #932: --stats-only dosen't work and open_images_downloader.py always not responding

shanhwa, 2021-02-19:
Dear Mr.Dusty,

I am following your steps to enter the command to execute:

$python3 open_images_download.py --class-names "vehicle" --data=data/vehicle
---------------
2021-02-19 14:49:51 - Download https://storage.googleapis.com/openimages/2018_04/class-descriptions-boxable.csv.
2021-02-19 14:49:52 - Requested 1 classes, found 1 classes
2021-02-19 14:49:52 - Download https://storage.googleapis.com/openimages/2018_04/train/train-annotations-bbox.csv.
---------------

This is the 5th time I tried, and it stopped here forever. I went to check the file in the directory. It quickly grew to 1.1GB, and it didn't increase anymore. After a few hours, I executed Ctrl+C to cancel, and it prompted:
---------------
File "open_images_downloader.py", line 139, in <module>
http_download(url, annotation_file)
File "open_images_downloader.py", line 75, in http_download
buf = f.read(1024)
File "usr/lib/python3.6/http/client.py", line 463, in read
n = self.readinto(b)
File "usr/lib/python3.6/http/client.py", line 507, in readinto
n = self.fp.readinto(b)
File "usr/lib/python3.6/socket.py", line 586, in readinto
return self._sock.recv_into(b)
File "usr/lib/python3.6/ssl.py", line 1012, in recv_into
return self.read(nbytes, buffer)
File "usr/lib/python3.6/ssl.py", line 874, in read
return self._sslobj.read(len, buffer)
File "usr/lib/python3.6/ssl.py", line 631, in read
return self._sslobj.read(len, buffer)
---------------

Then I execute again:
---------------
2021-02-19 15:50:06 - Requested 1 classes, found 1 classes
2021-02-19 15:50:06 - Read annotation file data/vehicle/train-annotations-bbox.csv
2021-02-19 15:57:41 - Available train images: 14756
2021-02-19 15:57:41 - Available train boxes: 48572

2021-02-19 15:57:41 - Download https://storage.googleapis.com/openimages/2018_04/validation/validation-annotations-bbox.csv.
2021-02-19 15:57:44 - Read annotation file data/vehicle/validation-annotations-bbox.csv.
2021-02-19 15:57:46 - Available validation images: 1434
2021-02-19 15:57:46 - Available validation boxes: 2070

2021-02-19 15:57:46 - Download https://storage.googleapis.com/openimages/2018_04/test/test-annotations-bbox.csv.
---------------

Then there is no response, I have been waiting for a long time.
I added the parameter --stats-only and got the same result

Looking forward your reply, Mr.Dusty.

--Your big fan.

dusty-nv, 2021-02-19:
Hi @shanhwa, it seems like you are having some network issue, unfortunately I cannot see a similar problem here.  Something you could try is manually downloading those .csv files (like `https://storage.googleapis.com/openimages/2018_04/test/test-annotations-bbox.csv`) to your dataset's directory. 

Then when you run the open_images_downloader.py tool, it will skip trying to download that large CSV file because it already found it on disk.

shanhwa, 2021-02-20:
Dear Mr.Dusty,

According to what you said, I tried to download these files manually, and then it worked normally, but it is worth mentioning that before this, the file existed, and in fact it was downloaded successfully, but it was stuck at the download stage.

BTW, after exporting the ONNX file, can the vehicle directory in ssd/data/vehicle be deleted?

dusty-nv, 2021-02-20:
Yes, it can be deleted. Although if you have a USB drive or something you may want to back it up in case you want it again and you have trouble downloading it.

shanhwa, 2021-02-20:
Thank you Mr.Dusty.

closed 2021-02-20 by shanhwa