dusty-nv/jetson-inference #933: Image Size for SSD Architecture

naimulhq, 2021-02-19:
When looking at the SSD Architecture, it seems like the input image needs to be 300x300x3. If we are using images much larger than that for training, will that increase the amount of time taken to train a model and would the large image size hurt accuracy?  If that's the case, would we have to resize the image and then annotate before training the model?

dusty-nv, 2021-02-19:
Hi @naimulhq, the images/annotations automatically get resized to 300x300 during the training process, so you don't have to worry about doing that.

naimulhq, 2021-02-19:
@dusty-nv If that is the cause, what can be some reasons as to why my model training takes longer than expected? In the "Retraining SSD Mobilenet" page, it mentions how the Nano can train 5 images per second. I was trying to train a model with approximately 700 images and only one epoch took me about 4 hours.

dusty-nv, 2021-02-19:
Larger images can take longer to load, yes, although it may be that you are low on memory.  That does seem long for one epoch. 
 Do you have Nano 2GB or Nano 4GB?  If you keep an eye on the memory/swap usage with `tegrastats`, what is it?

naimulhq, 2021-02-19:
I have a Nano 4GB. I will let you know what the tegrastats tell me in a bit but as of now, I have allocated 8GB of swap memory and disabled the GUI. Nevertheless, training time takes a long amount of time and there are instances where the worker just fails and throws an error.

dusty-nv, 2021-02-19:
Hmm.  You might want to try reducing the batch size and workers with `--batch_size=2 --workers=1`.   I wonder if the SD card is limiting the speed or is a slower SD card.

naimulhq, 2021-02-24:
@dusty-nv The link provided shows the SD card that I am using with the Jetson.[](https://www.amazon.com/Kingston-Digital-microSD-SDCX10-64GB/dp/B009X0G304/ref=sr_1_2?crid=N2RSIIJ8BJR6&dchild=1&keywords=kingston+sd+card+64gb&qid=1614142098&s=electronics&sprefix=kingston+sd+card%2Caps%2C505&sr=1-2)

naimulhq, 2021-03-01:
@dusty-nv I decided to opt to Google Colab for training purposes and everything ran smoothly.

closed 2021-03-01 by naimulhq

neilyoung, 2021-04-18:
@naimulhq Would you mind to share your Colab approach or provide some start pointers?

naimulhq, 2021-04-18:
@neilyoung Here is a link to the notebook that I made. You can use it for training. You might need to change some commands depending on what you are training. The notebook is designed for object detection models. Also, regular colab users have a 12 hour time limit. If you don't complete your training in 12 hours, you will automatically lose access to using their GPU resources.
https://github.com/naimulhq/NASACapstone/blob/main/JetsonTrainingNotebook.ipynb

naimulhq, 2021-04-18:
@naimulhq Also, I would check out this link. This is used to prevent idle timeouts from happenning.
https://stackoverflow.com/questions/54057011/google-colab-session-timeout

neilyoung, 2021-04-18:
@naimulhq Thank you very much for sharing. This looks exact like what I had in mind to build. Will definitely give it a try and report here. I already appeared that limitations and I would love to pay Google the fee to use the pro version but AFAIK they don't sell it outside the US

naimulhq, 2021-04-18:
@neilyoung Our group is using the Pro version which bumps us up to a max timeout of 24 hours. Eventually, our models became large enough that we could do a limited amount of epochs before our timeout. You could also look at AWS but I have no clue how to use that.

neilyoung, 2021-04-18:
@naimulhq A friend has used AWS for ML. It worked, but was pretty expensive.

neilyoung, 2021-04-19:
@naimulhq Hmm. Even the first few lines behave somehow strange on Colab (non-pro). Although I know, this is @dusty-nv 's "build from scratch" sequence, which forks fine on a Nano, I'm having here strange technical problems... 

The "build" folder is not created under "jetson-inference", as ordered, instead one level up, so that the following cmake fails.

Must be some Colab special, also because the folder names are preceded by "content/", what I have seen with other projects too already.

The sequence:

```
!sudo apt-get update
!sudo apt-get install git cmake libpython3-dev python3-numpy
!git clone --recursive https://github.com/dusty-nv/jetson-inference
!cd jetson-inference
!mkdir build
!cd build
!cmake ../
!make -j$(nproc)
!sudo make install
!sudo ldconfig
```

The log:

```
Ign:1 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  InRelease
Ign:2 https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/x86_64  InRelease
Get:3 https://cloud.r-project.org/bin/linux/ubuntu bionic-cran40/ InRelease [3,626 B]
Get:4 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release [697 B]
Hit:5 https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/x86_64  Release
Get:6 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release.gpg [836 B]
Get:7 http://ppa.launchpad.net/c2d4u.team/c2d4u4.0+/ubuntu bionic InRelease [15.9 kB]
Get:8 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Hit:10 http://archive.ubuntu.com/ubuntu bionic InRelease
Ign:11 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Packages
Get:11 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Packages [696 kB]
Get:12 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Hit:13 http://ppa.launchpad.net/cran/libgit2/ubuntu bionic InRelease
Get:14 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [2,112 kB]
Get:15 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu bionic InRelease [15.9 kB]
Get:16 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [31.4 kB]
Get:18 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [2,542 kB]
Get:19 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [24.5 kB]
Get:20 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [1,408 kB]
Get:21 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [395 kB]
Hit:22 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease
Get:23 http://ppa.launchpad.net/c2d4u.team/c2d4u4.0+/ubuntu bionic/main Sources [1,755 kB]
Get:24 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [425 kB]
Get:25 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [2,179 kB]
Get:26 http://ppa.launchpad.net/c2d4u.team/c2d4u4.0+/ubuntu bionic/main amd64 Packages [898 kB]
Get:27 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu bionic/main amd64 Packages [39.5 kB]
Fetched 12.8 MB in 4s (3,095 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.10.2-1ubuntu2.18.04.1).
git is already the newest version (1:2.17.1-1ubuntu0.8).
libpython3-dev is already the newest version (3.6.7-1~18.04).
Suggested packages:
  python-numpy-doc python3-nose python3-numpy-dbg
The following NEW packages will be installed:
  python3-numpy
0 upgraded, 1 newly installed, 0 to remove and 57 not upgraded.
Need to get 1,943 kB of archives.
After this operation, 10.9 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu bionic/main amd64 python3-numpy amd64 1:1.13.3-2ubuntu1 [1,943 kB]
Fetched 1,943 kB in 1s (1,444 kB/s)
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 1.)
[... 55 lines cut ...]
Submodule path 'python/training/detection': checked out '35987404f370f791581925934536768691d553c9'
Submodule 'ssd' (https://github.com/dusty-nv/pytorch-ssd) registered for path 'python/training/detection/ssd'
Cloning into '/content/jetson-inference/python/training/detection/ssd'...
remote: Enumerating objects: 908, done.        
remote: Counting objects: 100% (8/8), done.        
remote: Compressing objects: 100% (8/8), done.        
remote: Total 908 (delta 0), reused 1 (delta 0), pack-reused 900        
Receiving objects: 100% (908/908), 1.08 MiB | 2.95 MiB/s, done.
Resolving deltas: 100% (597/597), done.
Submodule path 'python/training/detection/ssd': checked out 'e7b5af50a157c50d3bab8f55089ce57c2c812f37'
Submodule path 'python/training/segmentation': checked out '6ae33ba3426665b8bde8923c0434d207ea4331e0'
Submodule path 'tools/camera-capture': checked out '1571f9048ff63ec5e5b523e4e41d1988905b9d84'
Submodule path 'utils': checked out 'c373f49cf21ad2cae7e4d7da7c41f4fd6473958f'
CMake Error: The source directory "/" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
/sbin/ldconfig.real: /usr/local/lib/python3.7/dist-packages/ideep4py/lib/libmkldnn.so.0 is not a symbolic link
```

The folder layout after this:

[image]

Any ideas?

neilyoung, 2021-04-19:
I think, this would be the proper command sequence (note the %cd)

```
!sudo apt-get update
!sudo apt-get install git cmake libpython3-dev python3-numpy
!git clone --recursive https://github.com/dusty-nv/jetson-inference
%cd jetson-inference
!mkdir build
%cd build
!cmake ../
!make -j$(nproc)
!sudo make install
!sudo ldconfig

```

neilyoung, 2021-04-19:
Well, wondering, how you ever have been able to compile the entire Dusty phalanx. It has so many references to Nvidia hardware...

I'm having moved forward a bit, but hit the next break:

```
/content/jetson-inference/utils/camera/gstCamera.cpp:37:10: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
          ^~~~~~~~~~~
```

And I understand that: This could have never been compiled on a Colab... 

I guess you were only using the Tensorflow python scripts and you also never managed to setup Dusty's environment on the Colab...

dusty-nv, 2021-04-19:
Yes, NvInfer.h is TensorRT.

The PyTorch training scripts are broken out into submodules, so you needn't clone the entire jetson-inference repo on x86.

https://github.com/dusty-nv/pytorch-ssd
https://github.com/dusty-nv/pytorch-classification

Those are the submodules under `jetson-inference/python/training`

neilyoung, 2021-04-19:
Yepp. Saw this. Thanks

naimulhq, 2021-04-19:
@neilyoung Ahhh I never noticed that. I was still able to train even with that typo but you are correct. It should be like that.

neilyoung, 2021-04-19:
Yepp. Since you only need the python modules and some dependencies there. I will provide you the link to my colab once I have "ported" Dusty's  "How to retrain mobilenet SSD" tutorial

neilyoung, 2021-04-19:
OK, here we have it. It is more or less (w/o the test) a transpose of Dusty's tutorial:

# Google Colab version of https://github.com/dusty-nv/jetson-inference/blob/dev/docs/pytorch-ssd.md
# Note: Make sure to instantiate your colab with a runtime of type GPU

I ran it once and it worked. Might be possible that I inserted a regression bug while assembling the steps.

```
# Prepare
!git clone https://github.com/dusty-nv/pytorch-ssd
%cd pytorch-ssd/
!wget https://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth -O models/mobilenet-v1-ssd-mp-0_675.pth
# Required in order to circumvent some ugly complains in the next step
!pip install urllib3==1.25.4 folium==0.2.1
!pip install -v -r requirements.txt

# Download images and annotations from Open Images Dataset
!python open_images_downloader.py --max-images=2500 --class-names "Apple,Orange,Banana,Strawberry,Grape,Pear,Pineapple,Watermelon"

# Train. Note: Change --epoch to something between 30 and 100
!python train_ssd.py --model-dir=models/fruit --batch-size=4 --num-epochs=1

# Export to ONNX
!python onnx_export.py --model-dir=models/fruit

# Download model
from google.colab import files
files.download('models/fruit/ssd-mobilenet.onnx') 

```

neilyoung, 2021-04-19:
OK, I managed to re-train the network to one person only. This at least was not achievable on the Nano for me. But there seems to be an invisible barrier with this approach. Same results: 45 fps one cam, 22 fps cam tow cams, 15 fps three cams.