dusty-nv/jetson-inference #934: detectnet doesn't work after I got the ssd-mobilenet.onnx

shanhwa, 2021-02-20:
Mr.Dusty,

I already have done:
$ python3 open_images_downloader.py --class-names "vehicle" --data=data/vehicle
$ python3 train_ssd.py --data=data/vehicle --model-dir=models/vehicle --batch-size=2 --epochs=5
$ python3 onnx_export.py --model-dir=models/vehicle

And when I am running the Live Camera Program with:
$detectnet --model=models/vehicle/ssd-mobilenet.onnx --labels=models/vehicle/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0

At this time, the camera is turned on, but I put the car model in front of the camera, it does not have any recognition frame.

dusty-nv, 2021-02-22:
Hi @shanhwa, it typically requires more than 5 epochs to fully train the model.   Did you try testing it on a static image with detectnet program?    Also, you can lower the detection threshold by specifying `--threshold=0.25` to detectNet (the default is 0.5).  You can make it lower until you get a detection.   In reality, if you need to keep lowering this, it means the model probably needs more training.

shanhwa, 2021-02-23:
Yes Mr.Dusty,

I have re-trained the model with:
$ python3 train_ssd.py --data=data/vehicle --model-dir=models/vehicle --batch-size=2 --epochs=10
(I changed the epochs to 10)
but if I don't use --threshold=0.25 parameter, It doesn't have any recognition frame, now the recognition rate only 25.2%

How to increase the recognition rate to 50%+

dusty-nv, 2021-02-23:
Generally you will want to keep training for more epochs.  For object detection, I will train it for 30 epochs minimum or 100 epochs if I want a good model.  

How many images are in your training dataset?

shanhwa, 2021-02-23:
I have 14756 images in my training dataset.

Mr.Dusty, when I finish this training, if I want to add some more images dataset, what can I do? Do I need to training all images again, or I just need to use execute this:
$ python3 train_ssd.py --data=data/<another direction>--model-dir=models/vehicle --batch-size=<x> --epochs=<x>

Thank you Mr.Dusty.

dusty-nv, 2021-02-23:
> when I finish this training, if I want to add some more images dataset, what can I do?

You can use the `--pretrained-ssd` argument and point it to the model checkpoint you want to resume from.

shanhwa, 2021-02-23:
Could you please give me the command completely?
like:
$ python3 train_ssd.py --data=data/(new images direction) --model-dir=models/vehicle --batch-size=1 --epochs=30 --pretrained-ssd
Is that right? Mr.Dusty.

dusty-nv, 2021-02-23:
You would want to look in your `models/vehicle` dir for the model name with the lowest loss (or the last epoch).   Then specify that filename to `--pretrained=<model-path>`  

You would also probably want to save the new models from your next training run to a new directory:

``` bash
$ python3 train_ssd.py --data=data/(new images direction) --model-dir=models/vehicle_run2 --batch-size=1 --epochs=30 --pretrained-ssd=models/vehicle/<best-model-filename>
```

Also, your new dataset should have images with the same classes as before.   Otherwise, the new dataset should also contain the old images too - otherwise the model will only recognize the new classes.

closed 2023-04-27 by dusty-nv