dusty-nv/jetson-inference #935: Integrating CUDA image manipulation with cpp tensorrt

OrcunCanDeniz, 2021-02-24:
I have a working TRT engine but I discovered preprocessing steps (resizing and cropping with opencv cpp) are lasting longer than forward prop itself. Do you have any examples or opinions on how can i integrate cuda resizing and cropping(or downsampling) of Cv::Mat type and feed it to TRT engine?  

Thanks in advance!

dusty-nv, 2021-02-24:
There are additional pre-processing functions meant for DNN's in `tensorConvert.h`:

https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorConvert.h

These do things like NCHW format conversion, mean pixel subtraction, pixel normalization, ect.

To use CUDA functions, you want to take the CPU memory from cv::Mat and copy it into GPU memory.  If you use my [`cudaAllocMapped()`](https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/cuda/cudaMappedMemory.h) function, you can just do a simple memcpy() since `cudaAllocMapped()` allocates shared CPU/GPU memory.   If you use `cudaMalloc()` to allocate the memory, you need to use `cudaMemcpy()`

OrcunCanDeniz, 2021-02-25:
Thanks, but what i need is actually something like [this](https://github.com/dusty-nv/jetson-utils/blob/master/cuda/cudaResize.cu). I dont have much knowledge about cuda, how can i port this one to my TRT code. Is it enough to perform allocations you mentioned?

closed 2023-04-27 by dusty-nv