dusty-nv/jetson-inference #936: Support for python 3.8

magnus-eigenvision, 2021-02-24:
Hi @dusty-nv 

As I understand it, jetson-inference currently only supports python 2.7 and 3.6. Python 2.7 reached end of life already and 3.6 will do so later this year.

I saw that TensorRT recently added support for python 3.8. Does this mean that we can expect support for python 3.8 in jetson-inference any time soon?

dusty-nv, 2021-02-24:
@magnus-eigenvision if you have the Python 3.7 dev packages installed before building jetson-inference, jetson-inference should detect them and build the jetson-inference extension modules for 3.7 as well.  

If you wanted to try 3.8, you could add it here:  https://github.com/dusty-nv/jetson-inference/blob/cf756333ab1876d90f107f54e7a3c5e58c3ddd5a/python/CMakeLists.txt#L8

Since I only use TensorRT from within C++, I don't depend on the TensorRT Python libraries.

Gensight, 2021-06-09:
When I try to make I get errors stemming from this
/usr/bin/ld: cannot find -ljetson-utils-python-38

Here's the rest of the errors:
collect2: error: ld returned 1 exit status
python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/build.make:206: recipe for target 'aarch64/lib/python/3.8/jetson_inference_python.so' failed
make[2]: *** [aarch64/lib/python/3.8/jetson_inference_python.so] Error 1
CMakeFiles/Makefile2:1245: recipe for target 'python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/all' failed
make[1]: *** [python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2021-06-09:
Hi @Gensight, can you post the full build log?  Thanks.

Gensight, 2021-06-12:
nano:~/Downloads/jetson-inference/build$ make -j$(nproc)
[ 40%] Built target jetson-utils
[ 41%] Built target camera-viewer
[ 43%] Built target video-viewer
[ 49%] Built target gl-display-test
[ 53%] Built target jetson-inference
[ 57%] Built target imagenet
[ 57%] Built target imagenet-console
[ 63%] Built target jetson-utils-python-27
[ 69%] Built target jetson-utils-python-36
[ 70%] Built target imagenet-camera
[ 71%] Built target detectnet-camera
[ 73%] Built target detectnet-console
[ 75%] Built target detectnet
[ 77%] Built target segnet
[ 79%] Built target segnet-camera
[ 79%] Automatic MOC for target camera-capture
[ 81%] Built target segnet-console
[ 81%] Built target camera-capture_autogen
[ 86%] Built target jetson-inference-python-27
[ 87%] Linking CXX shared library ../../aarch64/lib/python/3.8/jetson_inference_python.so
[ 91%] Built target jetson-inference-python-36
[ 96%] Built target camera-capture
/usr/bin/ld: cannot find -ljetson-utils-python-38
collect2: error: ld returned 1 exit status
python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/build.make:206: recipe for target 'aarch64/lib/python/3.8/jetson_inference_python.so' failed
make[2]: *** [aarch64/lib/python/3.8/jetson_inference_python.so] Error 1
CMakeFiles/Makefile2:1245: recipe for target 'python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/all' failed
make[1]: *** [python/bindings_python_3.8/CMakeFiles/jetson-inference-python-38.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2021-06-14:
Did you add `3.8` to here:  https://github.com/dusty-nv/jetson-inference/blob/cf756333ab1876d90f107f54e7a3c5e58c3ddd5a/python/CMakeLists.txt#L8

And do you have the dev packages installed for Python 3.8?

MDBox, 2021-10-08:
Just in case anyone else run into this issue. I had to also add 3.8 to another file in `jetson-utils` submodule.  

https://github.com/dusty-nv/jetson-utils/blob/b38357bbe33640613acb7616fd7e675adbeaab2a/python/CMakeLists.txt#L8

```
set(PYTHON_BINDING_VERSIONS 2.7 3.6 3.7 3.8)
```

```
Scanning dependencies of target jetson-inference-python-38
[ 96%] Linking CXX shared library ../../aarch64/lib/python/3.8/jetson_inference_python.so
[100%] Built target jetson-inference-python-38
```

franz6ko, 2022-11-01:
Libraries are being generated by default inside /opt/conda even if conda is not installed. I had to copy all the generated files to /urs/lib/python3.8/dist-packages/ to make it work.

Aldhanekaa, 2023-03-04:
> Just in case anyone else run into this issue. I had to also add 3.8 to another file in `jetson-utils` submodule. [...]

Where the `jetson-utils` submodule located? I already tried CMakeLists.txt in the jetson-inference foler but it seems like the  cannot find -ljetson-utils-python-38 still appearing

closed 2023-04-27 by dusty-nv

protonxaviernx, 2023-07-09:
hi 
i have a problem when i tried to install jetson-inference( i have xavier nx board)
when i write cmake ,,/ in the terminal, many problems appeared like:

    Skipping link https://files.pythonhosted.org/packages/be/3b/d16c9afacf5c6517f3b6cfbba7c53b46acabb0fab65d05af1aaa2160b189/Cython-0.28-cp36-cp36m-manylinux1_i686.whl#sha256=7abe6b8c7435f26ce022419d4a5bd5e81a9a31cc32d984939bb808709ecad528 (from https://pypi.org/simple/cython/); it is not compatible with this Python

    Skipping link https://files.pythonhosted.org/packages/77/1f/3bf78b16e7f4fc4bc17f005c5d6ca6cb8ab7a8cc94113a503aed909f6cbe/Cython-0.29.11-cp34-cp34m-manylinux1_i686.whl#sha256=bdf46b44b10bcd89b976f47a953e723d17bd1e16b71ca2b125e2f9833f464ac0 (from https://pypi.org/simple/cython/) (requires-python:>=2.6, !=3.0.*, !=3.1.*, !=3.2.*); it is not compatible with this Python

The dialog model downloader did not appear
thank you

dusty-nv, 2023-07-09:
@protonxaviernx you can ignore those, and the model downloader tool has been removed - the models are now automatically downloaded for you on-demand when you attempt to load them