dusty-nv/jetson-inference #937: Releasing GIL during inference

magnus-eigenvision, 2021-02-24:
When I use the python bindings of jetson-inference I expected the GIL to be released during calls to `detectNet.Detect()` and `imageNet.Classify()` etc. However, profiling suggests that the GIL is not released during these calls, though I may have missed something in my measurements.

Is it correct that the GIL is not released? If so, would it be easy to implement?

dusty-nv, 2021-02-24:
I haven't played around with this before, so I suspect you are correct that the GIL is not being released.

It looks like the `Py_BEGIN_ALLOW_THREADS` and `Py_END_ALLOW_THREADS` macros can be used to do it:
https://www.geeksforgeeks.org/releasing-gil-and-mixing-threads-from-c-and-python/

You could do a quick test by wrapping the detectNet::Detect() call in these macros here:
https://github.com/dusty-nv/jetson-inference/blob/cf756333ab1876d90f107f54e7a3c5e58c3ddd5a/python/bindings/PyDetectNet.cpp#L613

(remember to re-run `make` and `sudo make install` after making changes to the C/C++ code)

magnus-eigenvision, 2021-02-24:
Thanks, it works! I think this is relevant for many of the python bindings.

Sometimes I get the following error
```
[cuda]      device not ready (error 600) (hex 0x258)
[cuda]      /PATH/TO/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
```
but the program still seems to function properly. Maybe some extra call to `cudaDeviceSynchronize()` or similar is needed?

dusty-nv, 2021-02-24:
hmm.  you may want to try adding the synchronize before camera.Capture() or between camera.Capture() and net.Detect()

magnus-eigenvision, 2021-02-24:
To give you some context, I have one thread that gets a frame from a `videoSource` and calls `detectNet.Detect()`, and then puts the detections and a copy of the image in a python queue. The other thread gets the objects from the queue, does some postprocessing and generates an overlay on the frame, and then displays it on a `glDisplay`.

The problem was that I called `detectNet.GetNetworkFPS()` from the display thread. When I moved that call to the thread doing the detections, the errors disappeared.

Unfortunately, the threaded version didn't improve throughput for me, though. Detection takes ~60 ms. If I run everything in a single thread, `RenderOnce()` takes ~2 ms, but with the threaded implementation it takes ~40 ms. Probably there's some other shared resource that is blocking. Probably, detection and display are competing for the GPU.

dusty-nv, 2021-02-24:
Yea, adding additional threading for videoSource and videoOutput are unlikely to help matters, because videoSource is already threaded underneath for the capture part (and then it launches asynchronous CUDA kernels for the format conversion).   videoOutput is also mostly asynchronous - it returns before the video is actually rendered by the GPU (which is why RenderOnce returns in 2ms)

detectNet uses cudaDeviceSynchronize internally after the DNN is executed, because it needs to wait on the CUDA results to interpret/cluster the bounding boxes on CPU.

closed 2023-04-27 by dusty-nv