dusty-nv/jetson-inference #939: Latency problem on Jetson AGX Xavier when streaming using RTSP

AlexPuscasu, 2021-02-26:
Hello @dusty-nv ,

We are currently using the jetson-inference into an vehicle and pedestrian detection project.
As a short description we use our pre-trained TinyDSOD model with the _detectnet_ task to perform the detection.
The input is the video stream provided by some CCTV cameras which is delivered using the RTSP protocol. 
The cameras that we use are from Mobotix and Dahua with MJPEG and H264 encoding and the OBD is running on Jetson Nano, Jetson Xavier NX and Jetson AGX Xavier platforms.
The camera image has a 1280x720 resolution at 15-25fps and the model is running with 1024x640 resolution.

The problem we encounter is that we have a huge latency when performing the detection with a value of about 500ms on Jetson Nano and about 300ms on Xavier NX.
We also ran the OBD on AGX Xavier where we obtained the worst latency value of about 2 seconds. 

So far we tried the following possible solutions:
- setting the latency value to 0 (https://github.com/dusty-nv/jetson-utils/blob/833fc7998e34d852672277730a11aeed90024959/codec/gstDecoder.cpp#L521)
- profiling of the _detecnet_ task using the Nvidia Nsight Systems which revealed that there are some blocking points and the rtpjitterbuffer and udpsrc look like are full and hold down the execution. 
Another strange situation is that on the AGX there are no blocking points, but still the latency values are big despite it's more powerful then the other ones.

Here is the profiling results from the Jetson Xavier NX:
[image]
Here is the profiling results from the Jetson AGX Xavier:
[image]

Do you have any idea of where might be the problem that cause this huge latency value?
What other things I can try in order to solve it?

dusty-nv, 2021-02-26:
What is the TensorRT performance of your detectnet network that get reported in the console?

If you just use video-viewer application to view your camera, is there similar latency?

AlexPuscasu, 2021-03-01:
Hello,

I run the video-viewer and I obtained a latency of about 400ms(using H264) and 280ms(using MJPEG).
Here is also a log that I took when running the detectnet on Jetson Nano.
[log_nano.txt](https://github.com/dusty-nv/jetson-inference/files/6063229/log_nano.txt)

closed 2023-04-27 by dusty-nv