dusty-nv/jetson-inference #940: unable to build within the ROS noetic docker container

sardanian, 2021-02-27:
I am having an issue building jetson-inference within a ROS noetic docker container. I can build it on the native system (I am using an Xavier NX) but when I try and build it within the the ROS noetic docker container I am getting this error:

[ 69%] Linking CXX shared library aarch64/lib/libjetson-inference.so
/usr/bin/ld: cannot find -lnvcaffe_parser
collect2: error: ld returned 1 exit status
CMakeFiles/jetson-inference.dir/build.make:257: recipe for target 'aarch64/lib/libjetson-inference.so' failed
make[2]: *** [aarch64/lib/libjetson-inference.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Any idea on what I can do to fix this?

Thank you

dusty-nv, 2021-02-28:
Hi, you need to set you default docker runtime to nvidia so that it will find Jetpack libraries - https://github.com/dusty-nv/jetson-containers

Also regarding this specific library, see how in my dockerfile it is removed:   https://github.com/dusty-nv/jetson-inference/blob/0d53bcb884693f8c8bd7e22bb12dcd962c827a19/Dockerfile#L76

sardanian, 2021-03-02:
Great. This resolved my issue. I needed to execute:
sed -i 's/nvcaffe_parser/nvparsers/g' CMakeLists.txt

in my jetson-inference folder

thank you

closed 2021-03-02 by sardanian