dusty-nv/jetson-inference #942: Face Recognition using facenet.

apoorvmishra28, 2021-02-27:
Hi @dusty-nv , I'm trying to build face recognition system. So, facenet is working fine in detecting faces but I need some suggestion regarding the integration of face recognition alongside facenet. Can you please help me out with your suggestion.

dusty-nv, 2021-03-01:
Hi @apoorvmishra28, you might want to check out DLIB, I believe it does face recognition.   Essentially the features from the detected faces are stored in a database and compared against later to find matches.

closed 2021-03-13 by apoorvmishra28

fasa79, 2021-10-31:
Hi @apoorvmishra28 and @dusty-nv . I m still new here. Do you mind to share where can I find the DLIB and so I can learn to do the face recognition.