dusty-nv/jetson-inference #943: Trouble Retraining MobilnetSSD

vucic0001, 2021-03-01:
ief (Clamp Type):192,96,64::', 'Strain Relief (Screw Type):64,224,64::', 'T&B Sticker/Label Branding:192,224,64::', 'Unused Knockout Location:64,96,192::', 'Water Main Land Lug:192,96,192::', 'Westinghouse Sticker/Label Branding:64,224,192::', 'Wire Traveling Below Grade (Assumed Ground):192,224,192::')
2021-03-01 02:10:38 - Validation dataset size: 3
2021-03-01 02:10:38 - Build network.
2021-03-01 02:10:38 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2021-03-01 02:10:40 - Took 1.04 seconds to load the model.
2021-03-01 02:11:42 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2021-03-01 02:11:42 - Uses CosineAnnealingLR scheduler.
2021-03-01 02:11:43 - Start training from epoch 0.
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
warning - image e0000017 has object with unknown class '40A Breaker off position'
warning - image e0000002 has object with unknown class '5A Breaker off position'
Traceback (most recent call last):
  File "train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 989, in _next_data
    return self._process_data(data)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 1014, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.6/dist-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
IndexError: Caught IndexError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/worker.py", line 185, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataset.py", line 207, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 81, in __getitem__
    image, boxes, labels = self.transform(image, boxes, labels)
  File "/jetson-inference/python/training/detection/ssd/vision/ssd/data_preprocessing.py", line 34, in __call__
    return self.augment(img, boxes, labels)
  File "/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 55, in __call__
    img, boxes, labels = t(img, boxes, labels)
  File "/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 277, in __call__
    overlap = jaccard_numpy(boxes, rect)
  File "/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 30, in jaccard_numpy
    inter = intersect(box_a, box_b)
  File "/jetson-inference/python/training/detection/ssd/vision/transforms/transforms.py", line 13, in intersect
    max_xy = np.minimum(box_a[:, 2:], box_b[2:])
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

thought i had it right with the txt files but im not sure what im doing wrong

dusty-nv, 2021-03-01:
Hi @vucic0001, my guess is that your XML annotations have a malformed/corrupt bounding box in them.  What I recommend is uncommenting this line of code:

https://github.com/dusty-nv/pytorch-ssd/blob/master/vision/datasets/voc_dataset.py#L76

This will print out each ImageID and bounding box as it is loaded.  Also set the batch size to 1, and this will more easily let you see which image is causing the exception to occur.   If you can't get it to work, you can send me your dataset and I can take a look.

vucic0001, 2021-04-01:
it worked out i was able to retrain and have a stable model now thank you

closed 2021-04-01 by vucic0001