dusty-nv/jetson-inference #945: camera capture (input: images) problem

ingbeeedd, 2021-03-03:
I have an image directory as below, and I put a file option using the camera-capture program, but the files move quickly and then work like below, is there any way to annotate one by one in the file directory?

```bash
camera-capture file://images/
```

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed

[image]

[image]

dusty-nv, 2021-03-03:
Hi @ingbeeedd, please use [CVAT](https://cvat.org/) tool or similar for annotating images that you have already captured.

ingbeeedd, 2021-03-04:
Thank you for inform `CVAT`. I will close it

closed 2021-03-04 by ingbeeedd