dusty-nv/jetson-inference #946: Please help

tianzanaruk, 2021-03-04:
ifollowing Jetson AI Fundamentals - S3E4 - Object Detection Inference in youtube

first and i try to use 

./detectnet.py --network=ssd-mobilenet-v2 "images/humans_*.jpg" images/test/humans_%i.jpg

it's ok  but  when try to use 

./detectnet.py --network=facenet "images/humans_*.jpg" images/test/humans_%i.jpg

[TRT]    invalid engine cache file size (0 bytes)  networks/facenet-120/snapshot_iter_24000.caffemodel.1.1.7103.GPU.FP16.engine
[TRT]    detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network

Traceback (most recent call last):
  File "./detectnet.py", line 51, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network

thanks

dusty-nv, 2021-03-04:
Hi @tianzanaruk, try running `rm networks/facenet-120/snapshot_iter_24000.caffemodel.1.1.7103.GPU.FP16.engine` and then re-running detectnet

tianzanaruk, 2021-03-05:
thanks for help

closed 2021-03-28 by tianzanaruk