dusty-nv/jetson-inference #948: incremental re-training for object detection

shaojun, 2021-03-08:
Hi Dusty,
Thanks for the great effort for this great repo.
I'm re-trainning your SSD-Mobilenet model for **extend** it to recognize a new category of an object, which is successfully done by a PASCAL VOC format data I prepared.
I know the transfer learning above is based on the base model which already trainned by you on MS COCO dataset(over 60 categories), but I noticed the new re-trainned model above can only recognize the new single category, is that possible to make a `incrementally re-training`, that means the early MS COCO categories still can be recognized.

Thanks.

dusty-nv, 2021-03-08:
Hi @shaojun, the SSD-Mobilenet-v2-COCO and the SSD-Mobilenet-VOC (from pytorch-ssd) are from two different frameworks/models - the SSD-Mobilenet-COCO came from TensorFlow.   The models made by pytorch-ssd don't have COCO (rather VOC).

But let's say for example that you wanted to add a class to VOC using pytorch-ssd.   I haven't tried this, but from what I would gather it could work if your labels.txt included all the VOC classes, and then with your new class added onto the end.   Your custom dataset wouldn't actually contain the original VOC classes, but it would be necessary to include those VOC classes in your labels.txt so the model doesn't 'overwrite' them.   

Anyways, as I mentioned I haven't tried this, and from what I've read if you are adding multiple classes it is best to re-train on the entire cumulative dataset (old + new)

shaojun, 2021-03-08:
Ok, i may get some misunderstanding.
As you suggested, for recognize all classes in this repo's default detectNet sample ( default model trainned on MS COCO) and new one from me( image dataset is VOC format),  i need convert MS COCO dataset to VOC, then merge in my new class dataset(jpeg, annotation, labels), finally form a new dataset, then using transfer learning based on  mobilenet-v1-ssd-mp-0_675.pth (which pretrained on PASCAL VOC dataset), am i correct?

dusty-nv, 2021-03-08:
That is essentially correct.   It may be difficult to convert MS COCO to Pascal VOC format, I'm not sure.  It may be easier to write/use a COCO dataloader for PyTorch and integrate that with pytorch-ssd.  Also training the PyTorch model for MS COCO will probably take awhile because it is large dataset.

umarfm, 2022-06-16:
So I wanted to train a model for a 'person' class (already in the pre-trained classes) and a 'box' class (which is new to the model). After labeling my data for both classes, and setting it up for training with pre-trained weights as the start point, the person class was detectable after the first 15 epochs as it already had pre-trained weights whereas the box class had to wait for another 2-300 epochs to be detectable with a good accuracy.

closed 2023-04-27 by dusty-nv