dusty-nv/jetson-inference #951: Re-training SSD-Mobilenet

huansu, 2021-03-09:
I followed the steps of the YouTube tutorial to execute the command, but I got this issue, who can help me?
`Traceback (most recent call last):
  File "train_ssd.py", line 243, in <module>
    target_transform=target_transform, is_test=True)
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 33, in __init__
    raise IOError("missing ImageSet file {:s}".format(image_sets_file))
TypeError: unsupported format string passed to PosixPath.__format__
`
(Jetson nano Developer Kit)

dusty-nv, 2021-03-10:
Hi @huansu, is this from a dataset that you collected yourself?

Can you check which image set files you have under `ImageSets/Main`?   There should be test.txt, train.txt, trainval.txt, val.txt

if you are missing any of those, copy them from train.txt for example

huansu, 2021-03-10:
@dusty-nv , thanks. I solved the problem with your suggestion.
And I got a new one, I tried to change the code to solve it, but failed.
`/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2021-03-10 18:48:44 - Epoch: 0, Step: 10/28, Avg Loss: 10.0993, Avg Regression Loss 3.7271, Avg Classification Loss: 6.3722
2021-03-10 18:48:49 - Epoch: 0, Step: 20/28, Avg Loss: 6.6907, Avg Regression Loss 2.7819, Avg Classification Loss: 3.9088
Traceback (most recent call last):
  File "train_ssd.py", line 346, in <module>
    val_loss, val_regression_loss, val_classification_loss = test(val_loader, net, criterion, DEVICE)
  File "train_ssd.py", line 165, in test
    return running_loss / num, running_regression_loss / num, running_classification_loss / num
ZeroDivisionError: float division by zero
`

huansu, 2021-03-11:
I made a wrong dataset by myself.

closed 2021-03-11 by huansu

agarces7, 2021-09-23:
@dusty-nv tengo el mismo error....estoy realizando con mi propio conjunto de datos y tengo la misma respuesta

Puliesi, 2024-10-28:
Estou com o mesmo problema:
ZeroDivisionError: float division by zero