dusty-nv/jetson-inference #952: HI, I build jetson-utils in x86 PC machine, and there are some question.

lzcchl, 2021-03-10:
I have build jetson-utils in x86 PC machine because API is great, and it works.

but I use API like 'cudaRGB8ToBGR8' or 'cudaNV12ToRGB' and so on, I find glDisplay Render cost more time than before(50ms), 
I guess that uchar3* is push to GPU from CPU when 'cudaRGB8ToBGR8', but glDisplay Render pull uchar3* to CPU from GPU, is that any API to download uchar3* fast from GPU to CPU...

or, have you use jetson-utils on x86 PC.

lzcchl, 2021-03-10:
emmm... if I remove 'cudaRGB8ToBGR8' operation, glDisplay Render cost 20ms in x86 PC(1080P)

lzcchl, 2021-03-10:
emmm... if I remove 'cudaRGB8ToBGR8' operation, glDisplay Render cost 20ms in x86 PC(1080P), but on jetson nx, glDisplay Render only cost 1~5ms

dusty-nv, 2021-03-10:
Hi @lzcchl, I haven't used jetson-utils on PC, but my guess is the slowdown is happening because I use CPU/GPU shared memory on Jetson (and on PC that memory would continually be transferred over PC).   

Here is how I allocate all my CPU/GPU memory on Jetson:  https://github.com/dusty-nv/jetson-utils/blob/1f3709f48258c2d75500c35605e8f6f4a3447afc/cuda/cudaMappedMemory.h

Instead you may want replace the internals of that function so that it uses cudaMallocManaged() - https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__MEMORY.html#group__CUDART__MEMORY_1gd228014f19cc0975ebe3e0dd2af6dd1b

I am not sure if it will work without issue, but I suspect that is what's happening.
The alternative would to be explicitly use cudaMalloc() / cudaMemcpy()

lzcchl, 2021-03-11:
So it is. I used ‘cudaMappedMemory.h’ in my code yesterday. just like :
{
uchar3* frame = NULL;
cudaAllocMapped(&frame, inputW, inputH);
...
...
CUDA_FREE_HOST(frame);
}

 I thought that if I use it, there will be shared memory in x86 PC machine environment, but my attempt yesterday told me that it doesn't seem to be the case.

I think shared memory is the key to ‘jetson-utils’, it great on jetson platform.

dusty-nv, 2021-03-11:
The secret is that shared memory on Jetson uses the same physical DRAM, because the Jetson's CPU and integrated GPU shares the same physical memory.  So there are no memory copies or transfers.

However on PC, the discrete GPUs have their own memory separate from CPU, so when using `cudaAllocMapped` that memory is read over PCIe each time.   So it is certainly optimized for the Jetson use-case.

closed 2021-03-12 by lzcchl