dusty-nv/jetson-inference #953: about output rtmp run error

lzcchl, 2021-03-11:
hello, I find gstEncoder support rtmp stream outpu in https://github.com/dusty-nv/jetson-utils/blob/1f3709f48258c2d75500c35605e8f6f4a3447afc/codec/gstEncoder.cpp, it's exciting! but when I run, it get error like this:
{
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! rtmpsink location=rtmp://192.168.1.11 
[gstreamer] gstEncoder -- failed to create pipeline
[gstreamer]    (could not link rtph264pay0 to rtmpsink0)
[gstreamer] gstEncoder -- failed to create encoder engine
}

emm..., my jetson ip is 192.168.1.11, so I set rtmp address is "rtmp://192.168.1.11:8554/live/xxx", and my test is "./build/aarch64/bin/video-viewer /dev/video0 rtmp://192.168.1.11:8554/live/xxx", what should I do?

and I test rtp is work, the message like this, just run command "./build/aarch64/bin/video-viewer /dev/video0 rtp://192.168.1.11:8554/live/xxx", only change rtmp to rtp:
{
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! udpsink host=192.168.1.11 port=8554 auto-multicast=true
[video]  created gstEncoder from rtp://192.168.1.11:8554/live/xxx
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://192.168.1.11:8554/live/xxx
     - protocol:  rtp
     - location:  192.168.1.11
     - port:      8554
  -- deviceType: ip
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
}

lzcchl, 2021-03-11:
rtmp support is exciting, hope rtsp support .. :-)

lzcchl, 2021-03-11:
nx@nx-desktop:~$ gst-inspect-1.0 | grep 264
x264:  x264enc: x264enc
videoparsersbad:  h264parse: H.264 parser
libav:  avmux_ipod: libav iPod H.264 MP4 (MPEG-4 Part 14) muxer
libav:  avdec_h264: libav H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 decoder
libav:  avenc_h264_omx: libav OpenMAX IL H.264 video encoder encoder
omx:  omxh264enc: OpenMAX H.264 Video Encoder
omx:  omxh264dec: OpenMAX H.264 Video Decoder
nvvideo4linux2:  nvv4l2h264enc: V4L2 H.264 Encoder
rtp:  rtph264pay: RTP H264 payloader
rtp:  rtph264depay: RTP H264 depayloader
typefindfunctions: video/x-h264: h264, x264, 264
uvch264:  uvch264src: UVC H264 Source
uvch264:  uvch264mjpgdemux: UVC H264 MJPG Demuxer

nx@nx-desktop:~$ gst-inspect-1.0 | grep 265
videoparsersbad:  h265parse: H.265 parser
libav:  avdec_h265: libav HEVC (High Efficiency Video Coding) decoder
omx:  omxh265enc: OpenMAX H.265 Video Encoder
omx:  omxh265dec: OpenMAX H.265 Video Decoder
nvvideo4linux2:  nvv4l2h265enc: V4L2 H.265 Encoder
x265:  x265enc: x265enc
de265:  libde265dec: HEVC/H.265 decoder
rtp:  rtph265pay: RTP H265 payloader
rtp:  rtph265depay: RTP H265 depayloader
typefindfunctions: video/x-h265: h265, x265, 265

dusty-nv, 2021-03-11:
Hi @lzcchl, see this post:  https://github.com/dusty-nv/jetson-utils/issues/53#issuecomment-788455778

I just patched this in https://github.com/dusty-nv/jetson-utils/commit/ebf55857d60ba0be5c655cacf8eb8e9348ace714

So updated/re-clone your jetson-inference repo and try again please.

lzcchl, 2021-03-12:
emm... I had re-clone jetson-inference repo and try again, the error "could not link rtph264pay0 to rtmpsink0" be solved, however   there is a new bug, that seems like "https://github.com/dusty-nv/jetson-utils/issues/56", pipeline fll, skipping frame.

I just rebuild jetson-inference, and run example with command "./video-viewer /dev/video0 rtmp://192.168.1.11:8554/live/xxx"

lzcchl, 2021-03-12:
emm... I had re-clone jetson-inference repo and try again, the error "could not link rtph264pay0 to rtmpsink0" be solved, however there is a new bug, that seems like "dusty-nv/jetson-utils#56", pipeline full, skipping frame.

I just rebuild jetson-inference, and run example with command "./video-viewer /dev/video0 rtmp://192.168.1.11:8554/live/xxx"

lzcchl, 2021-03-12:
[image]

closed 2023-04-27 by dusty-nv