dusty-nv/jetson-inference #954: Re-training SSD-model

huansu, 2021-03-12:
@dusty-nv Hi, Thanks for your great job firstly.
I have a question that need your help. I want to do some changes on the bounding box, what should I do?
Where I can find the code to change?

dusty-nv, 2021-03-12:
Hi @huansu, do you mean modifying the bounding boxes during training, or during inference.

As @themannoshoes pointed out, the inferencing code is in `jetson-inference/c/detectNet.cu`

During training, I think it depends on which type of dataset you are using.  For example, if using Pascal VOC type dataset, the code that parses the bounding boxes is found here:  https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/datasets/voc_dataset.py#L139

huansu, 2021-03-12:
@dusty-nv Thanks for your reply.I want to modify the bounding boxes  during inference. I will have a shot at it.
 @themannoshoes Thank you.

huansu, 2021-03-12:
@dusty-nv thanks, but I want to use python to run during inference, what should I do? Where I could find the *.py to run as <detectnet --model=$NET/ssd mobilenet.onnx --labels=$NET/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0>. Where should I change the code with bbox during inference?

dusty-nv, 2021-03-12:
You should just be able to edit the members of the detectNet.Detection objects.  See this page:

https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#detectNet

Scroll down for the jetson.inference.detectNet.Detection members

closed 2022-02-23 by huansu